photobooth-app/photobooth-app #272: Liveview Canon EOS750D

ruhlandm1, 2024-05-26:
**Describe the bug**
I´m able to capture a photo via gphoto2 from the Canon EOS750D - but I don´t get a live preview. 

**To Reproduce**
Steps to reproduce the behavior:
2. Click on Image
3. Get Photo but no live preview

**Expected behavior**
Live preview from the camera

**Additional context**
Camera is connected via usb; ist putted on manual focus; tried different modes (auto, video, etc.) on the camera itslef; no changes;

ruhlandm1, 2024-05-26:
Ok, I´ve solved it by my own: 

```
systemctl --user stop photobooth-app.service
gphoto2 --camera="Canon EOS 750D" --set-config viewfinder=1
systemctl --user start photobooth-app.service
```

mgineer85, 2024-05-26:
Hey, my camera 1100d switches to viewfinder mode automatically and instead I need to disable viewfinder before capture: https://github.com/photobooth-app/photobooth-app/blob/29396c59dd9e5a22ac108a76839280d126ec6ea1/photobooth/services/backends/gphoto2.py#L297

You need to set to one once before you start the app? Liveview still works after first picture captured?

ruhlandm1, 2024-05-26:
No works fine; after the photo it´s just sending a new livefeed

closed 2024-05-26 by mgineer85