photobooth-app/photobooth-app #274: Raspberry shuts itslef down

ruhlandm1, 2024-05-26:
**Describe the bug**
Raspberry Pi4+ runs without any Problems, as long as the Photobooth APP is running, but no camera is connected. As soon as the Camera is connected it takes between 1-3 minutes and the pi is no longer reachable. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start Photobooth
2. Connect Camera
3. Wait for 1-3 Minutes
4. See error

**Expected behavior**
Should be reachable without any problems. 

I´ve attached the logfile; can someone help me?

[photobooth_20240526.log](https://github.com/photobooth-app/photobooth-app/files/15447922/photobooth_20240526.log)

mgineer85, 2024-05-26:
There is in current version some issue that causes a high cpu load when the connection to the camera breaks. 
Maybe this is causing the RPI to be inresponsive because the webserver answers only very very slow then.

ruhlandm1, 2024-05-26:
hmm - is there a older version that hasn´t had this issue? Or is there a workaround? Do you know why the connection breaks? Is there something we could do about that?

mgineer85, 2024-05-26:
Actually, the high cpu load is a problem that occurs after the connection broke. So if the connection is OK, the cpu won't see high load.

Please ensure the cabling is right and rebooth the whole system as well as restart the camera. Do not enable RAW files in the camera helps also. Do not enable the AUTO mode but just a bit more manual as there seem to be issues with gphoto2 and auto mode.

mgineer85, 2024-05-26:
changes have been made here, but it really only helps if things gone wrong before already: https://github.com/photobooth-app/photobooth-app/commit/9f7e23439f49d1e6138f772c054fdd8c720a9fa9

mgineer85, 2024-05-26:
You could update to the latest version that is released as a beta now: https://pypi.org/project/photobooth-app/3.1b0/

ruhlandm1, 2024-05-27:
thank you very much for your fast response; I found out that disabling the option "Interrupt liveview for Photo" (or something similar - do not have the exact name in mind) helps; After that everything worked; Anyway, I´ll try to update to the beta version;

ruhlandm1, 2024-05-27:
ok I´ve justed updated to the version you´ve created - so fare everything works really fine; the camera gets recognized and everything is running stable (at least for the last hour); 

the only lasting mistake was that the selfstarting of the service after a reboot; I run the pi headless by the way; I solved it with a cronjob and the following line: 

@reboot systemctl --user start photobooth-app.service

mgineer85, 2024-05-28:
> too earlie - as soon as i close putty the service ends; does this happen because there is no screen connected to the pi?

No need for a screen. What happens?
There is no need for the cronjob usually it will start automatically on boot.

mgineer85, 2024-05-28:
Maybe the app shuts down because it was opened in the console?
Regular version 3.1.0 is out, so maybe you can update to that version.

ruhlandm1, 2024-05-31:
I´ve just updated to the latest version. Even got on of the Canon Selphy Printers to work. What I found out: 

1. App is running stable as long as you aren´t connected via any ssh console
2. App is running stable as long as you are connected to the gui with just one console
3. App is running stable as long as you use a capable enddevice (used some old tablets and the had a big delay [showed countdwon number 2 from 5 but picture was already taken])

Just one last question - I´m no longer able to login to the admin page [wrong password] - have you changed anything about that? Or ist the password stored somwehere in the config?

mgineer85, 2024-05-31:
Yes, to avoid users accessing the system there is a password now: https://github.com/photobooth-app/photobooth-app/releases/tag/v3.1.0
Default is 0000

mgineer85, 2024-05-31:
There is a setting in backends called livestream_framerate. For old devices you could lower the number , maybe it works for them. Otherwise the device is not capable to run the app.

closed 2024-06-04 by mgineer85