photobooth-app/photobooth-app #281: Admin password not stored correctly after saving - 3.1.0 fresh install

PixelAlarm, 2024-06-03:
**Describe the bug**
I think there is a bug in the authentication for admin panel.

On first login, the default password works.
When saving the common page (without editing password), the default (plain text) changes to stars (12 digits).
In the config.cfg the 12 stars are stored/displayed.

When logging in next time, the default passord will not work anymore.
Typing in 12 stars will work.

So I thing the encryption of the password isn't working.

**To Reproduce**
Steps to reproduce the behavior:
1. Log in first time
2. Go to admin Panel > common
3. Save page without changing password
4. Log out and try to log in again

**Offtopic**
I'm blown away by your work. Keep it up.
I'm so happy to have moved to your solution from another.

mgineer85, 2024-06-04:
Hi, thank you!

You are on the latest 3.1.0, right?
The ***** is to hide the actual password to the user. The actual password is emitted in the admin-config rest api only after authentication.

So, if you login in the config tab you should see the clear text password (0000). If there are **** maybe the browser still has some caching of the old frontend, F5 could help even though I think it should automatically invalidate the cache.

I installed a version from pip and tried right now. It works for me ;)   
Which browser do you use?

PixelAlarm, 2024-06-04:
Hi, yes I am on the newest 3.1.0.

I think I found the problem.
When hitting "save" twice, it will actually save the 12 stars as new password and override the password that was set.
Tried it with safari, firefox and chromium.

Maybe you can check your behavior when clicking the save button twice :)

mgineer85, 2024-06-05:
Found the issue. After save the config is reloaded from the server and that is using the api endpoint from initialization where no credentials are send to the server.
It's corrected and will be fixed in next release.

closed 2024-06-05 by mgineer85