photobooth-app/photobooth-app #282: Collage: Canvas Fill Background will cover Canvas Img Background.

PixelAlarm, 2024-06-03:
**Is your feature request related to a problem? Please describe.**
I'm trying to use a solid color as background in collage and want to overlay a texture as .png.
This way I could reuse a texture and be able to change the background color on the device with the color picker.

Problem:
I think the solid color is applied on top of my texture. The texture is not visible in the processed collage.
So the Img and Fill should be switched. (Fill would always cover the Img, because it's solid.)

**Describe the solution you'd like**
Be able to set a color as background and have my transparent texture laying on top of it.

**Describe alternatives you've considered**
A workaround that I configured, but that isn't as intuitive:
Add a predefined image as first image and stretch it to the collage size.

**Additional context**
Maybe I'm doing something wrong.
It's not a dealbreaker, as there is s workaround.

mgineer85, 2024-06-04:
During implementation I considered to not use both of the options at the same time actually.
Could you send me the material, so I can test with your actual images?
Not sure if it's as easy as changing the sequence in the application or other things may break then ;)

PixelAlarm, 2024-06-04:
Hi, sure.
[image]

mgineer85, 2024-06-04:
Oh I see how that could give a nice finish. 
The reason I implemented as is was with respect to the chromakey removal.
Let me check if there would be combinations that could break.

mgineer85, 2024-06-05:
Should be fixed in next version.

The image was paletted and transparency saved to the image but not in an alpha channel. So it failed to detect the transparency during the processing and ignored the file to just continue. There should have been errors in the log about this.

mgineer85, 2024-06-05:
Regarding the solid that covers the image I have no good solution yet.

PixelAlarm, 2024-06-11:
Hi Michael,

I read a bit into the code and an easy fix could be to switch "_apply_stage_fill_background" and "_apply_stage_img_background" in mediaprocessingservice.py.

Now the stages are:
1. Transparent picture
2. Solid color (fill background) behind transparent picture
3. transparent (img background) behind solid color (fill background) behind transparent picture

--> Solid "fill background" will now always cover any transparent/nontransparent "img background".

New stages would be:
1. Transparent picture
2. Transparent (img background) behind transparent picture
3. Solid color (fill background) behind transparent (img background) behind transparent picture

Could you look into this?
Will try to test it myself.

mgineer85, 2024-06-12:
I did just change it during testing but it did not have the desired effect (for me) . Maybe when you test it’s different because you use different images to test.

PixelAlarm, 2024-06-12:
For me changing the order for Collage as stated above had indeed the desired effect.

Used this png as img:
https://www.vecteezy.com/png/41643335-paper-wrinkled-and-crumped-texture-transparent-glued-poster-template

Should I create a pull request or can you change the order based on my info?

Best regards

mgineer85, 2024-06-13:
Ok, I will check again.

mgineer85, 2024-06-16:
Without actual images of the composition showing the problem I am not sure to understand correctly.
Also not sure what to achieve when you say you do have a transparent image.
Is the transparent image made from the greenscreen removal?

Following is, when the stages are reordered like you suggested above.
The green color is canvas fill, blue is fill per capture. The persons got removed because the greenscreen removal is applied to them and the blue fill as well as the sprinkles shine through. 
[image]

In an ideal world: How would you embed this in your photos? Like in which sequence you would create the layers in photoshop? https://www.vecteezy.com/png/41643335-paper-wrinkled-and-crumped-texture-transparent-glued-poster-template

PixelAlarm, 2024-06-16:
I try to explain my use case.
So I would like to have some generic texture pngs to give the solid background a nice finish (e.g. imitate colored paper).
I would then be able to change the background color with the color picker and have the paper texture in between the color and the pictures.

With the actual implementation, the texture is fully covered by the solid background:
[image]

With changed order as described, I get this result:
[image]

mgineer85, 2024-06-17:
Ok, get that. I never used it without the front image that would cover the solid color anyways again.
I think we can make that work.

mgineer85, 2024-06-17:
Will be available in next release. 
I think the processing need to be revised to something like this in the future: https://github.com/dkraczkowski/dkraczkowski.github.io/tree/main/articles/crafting-data-processing-pipeline

but for now it should be ok.

commit 2e6b2b48 (2024-06-05): fix #282
commit d8304e87 (2024-06-17): fix #282
closed 2024-06-17 by mgineer85