photobooth-app/photobooth-app #288: How to choose from multiple printer configurations on result screen / gallery

mgineer85, 2024-06-06:
### Discussed in https://github.com/photobooth-app/photobooth-app/discussions/283

<div type='discussions-op-text'>

<sup>Originally posted by **PixelAlarm** June  4, 2024</sup>
Hi,

Offtopic:
I have a printer that prints 20x15cm. Possible media formats are 1x full-size: 20x15cm or 2x half-size: 10x15cm.

Actual question:
My idea was to have two printer routines: 1x full-size or 2x half-size.
Now I set up two routines, but I cannot choose between configurations on result screen or from gallery.
There is only one button that will use the first configuration.

Is there a way to choose between the routines?</div>

mgineer85, 2024-06-07:
https://github.com/photobooth-app/photobooth-app/discussions/283#discussioncomment-9689024

mgineer85, 2024-06-07:
https://github.com/photobooth-app/photobooth-app/issues/276

mgineer85, 2024-06-07:
[image]
@PixelAlarm @baumrasen Just a mockup, it could look like this

PixelAlarm, 2024-06-07:
Looking forward to the possibility :)

For my taste it would be nice to have only 1 print/share bottom at the top and if pressed there would be a floating option window with icons and short description of the options. (similar to the bottom menu of front screen)

mgineer85, 2024-06-07:
Yes, I would like to avoid cluttering the screen too much with distracting stuff.
I think it could be configurable how many items are displayed and which elements are hidden in a dropdown.

mgineer85, 2024-07-14:
[image]
[image]

This feature will be included in next release.
QR share config and printing config will be lost because the config got rearranged - so it will be released as v4 to signal breaking change.

mgineer85, 2024-07-14:
This will be included in v4 - beta1 is released: https://pypi.org/project/photobooth-app/4.0.0b1/

It's major version change because qrcode share and printing settings will be lost after update :|

closed 2024-07-14 by mgineer85