photobooth-app/photobooth-app #289: Adjust material icon weight via private.css

PixelAlarm, 2024-06-07:
**Is your feature request related to a problem? Please describe.**
I would like to alter the icon weight of the action buttons, since they are pretty bold.
For material icons it is possible to adjust the font-weight.

The text-weight is actually adjustable via private.css with:
.action-button {
    font-weight: 200;
}

Altering the font-weight for the icon does nothing.

**Describe the solution you'd like**
Use this styling:

.action-button *>.q-icon {
    font-weight: 200;
}

PixelAlarm, 2024-06-07:
I just experimented a bit.
Seems actually to be able to make the icons more bold but not lighter.

In https://fonts.google.com/icons the icons can be made much lighter.

mgineer85, 2024-06-07:
This is a limitation of the material icons. They are included in 400 weight only:
https://quasar.dev/options/installing-icon-libraries/
They are part of the extras package.
https://github.com/quasarframework/quasar/tree/dev/extras

Not sure if it is possible to add a custom icon set to the private.css and use that.

PixelAlarm, 2024-06-15:
I can't find a way to use other icons than the implemented ones.
I'm a bit sad, since smaller weight would give the frontend a nice look in my opinion.

Would it be possible to use pngs / svgs instead of the hardcoded icons?
Inserting pngs via html didn't work out.

It would also be really nice to adjust the form of the buttons.
For me, round icons would really look nice, but adding the "round;" parameter to the action buttons does nothing.

PixelAlarm, 2024-07-02:
I found a solution:
https://stackoverflow.com/questions/73238898/why-does-opera-display-googles-material-symbols-outlined-icon-font-incorrectl

**Adding this to private.css will let you change the weight:**

@font-face {
  font-family: 'Material Symbols Outlined';
  font-style: normal;
  font-weight: 300 700;
  src: url(https://fonts.gstatic.com/s/materialsymbolsoutlined/v39/kJEhBvYX7BgnkSrUwT8OhrdQw4oELdPIeeII9v6oFsI.woff2) format('woff2');
}

.material-symbols-outlined {
  font-family: 'Material Symbols Outlined';
  font-weight: normal;
  font-style: normal;
  font-size: 24px;
  line-height: 1;
  letter-spacing: normal;
  text-transform: none;
  display: inline-block;
  white-space: nowrap;
  word-wrap: normal;
  direction: ltr;
  -webkit-font-feature-settings: 'liga';
  -webkit-font-smoothing: antialiased;
}

.action-button *>.q-icon {
	font-family: 'Material Symbols Outlined';
	font-weight: 300 !important;
}

mgineer85, 2024-07-03:
Hello @PixelAlarm , thank you for digging into this!

This way I found out that google has two icon sets nowadays: Material Icons and Material Symbols.
It seems Quasar (our frontend framework) supports both https://quasar.dev/options/quasar-icon-sets/ it seems and Material Symbols is considered as the new version of the library.

The new version supports several weights natively https://fonts.google.com/icons?icon.set=Material+Symbols

Another reference for supported fonts:
https://github.com/quasarframework/quasar/tree/dev/extras

I keep it open as enhancement and will check if the new lib is just a dropin replacement.

PixelAlarm, 2024-07-04:
Thanks for your reply.

Couldn‘t it be as easy as adding the line to the extras section in photobooth-frontend/quasar.config.ts?

extras: [
      'material-symbols-outlined', // use it with o_sym_ prefix
      'material-icons', // optional, you are not bound to it
      'material-icons-outlined', // optional, you are not bound to it use with o_ prefix
    ],

mgineer85, 2024-07-04:
Likely. But I’d like to remove the old lib then to avoid adding too much unused stuff. And that might need changing all Icon references.
But it should be easy.

mgineer85, 2024-07-05:
Material Icons:
[image]

Material Symbols:
[image]

Weight can be changed easily now:
[Video.webm](https://github.com/photobooth-app/photobooth-app/assets/3868450/b9a248b0-c6f2-48c8-a573-cf4dfd9d0c9c)

The new icons need sym_o_ prefix usually. It will be added automatically to the icons defined in the configuration.
The updated UI will be included in next release.

mgineer85, 2024-07-12:
This will be included in the next release, so I am closing it here.
You probably need to adjust your custom css again after updating.

closed 2024-07-12 by mgineer85

PixelAlarm, 2024-07-15:
Hi, just tested with 4.0.0b1.

Should it be as easy as changing the private.css?
This doesn't work for me:

`.action-button *>.q-icon {
	font-weight: 100;
}`

My old workaround still works though :)

mgineer85, 2024-07-15:
I am very sorry, there has been something wrong with the commits.
It will work in next beta as intended and shown in the video:

[Video.webm](https://github.com/photobooth-app/photobooth-app/assets/3868450/b9a248b0-c6f2-48c8-a573-cf4dfd9d0c9c)