photobooth-app/photobooth-app #291: There is no current event loop in thread 'AnyIO worker thread'. / HTTP 500

harakaio, 2024-06-08:
**Describe the bug**
Hey together,
I installed the photobooth-app through the fantastic documentation. That is really helpful!
Unfurtonately I'm not able to make pictures with my Canon EOS 1100D.
The livestream via webcam does work.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Photobooth-Project via pipx
2. make picture
3. error mesasge :/ ==> 

08.Jun.24 21:28:17	HTTPException: HTTPException(status_code=500, detail="something went wrong, Exception: There is no current event loop in thread 'AnyIO worker thread'.")
08.Jun.24 21:28:17	There is no current event loop in thread 'AnyIO worker thread'.
	08.Jun.24 21:28:17	There is no current event loop in thread 'AnyIO worker thread'.

**Expected behavior**
The photobooth to make a picture would be great.

**Screenshots**

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit installed on 08.06.2024 via online image from RPi Imager
 - Operating System: Debian GNU/Linux 12 (bookworm)
 - Kernel: Linux 6.6.31+rpt-rpi-v8

mgineer85, 2024-06-09:
Which installation method did you use on the system?

mgineer85, 2024-06-09:
Ah, doesnt matter, found the issue in the updated state-machine package. Same experience on windows which I use for dev.

Workaround (pinning statemachine package) fix is on the way.

closed 2024-06-09 by mgineer85

mgineer85, 2024-06-09:
Updated version released.    I think it should fix it for you. If not, check which python-statemachine version is installed. It has to be 2.2 for now, not the latest 2.3

https://github.com/photobooth-app/photobooth-app/releases/tag/v3.1.4

harakaio, 2024-06-09:
> Updated version released.    I think it should fix it for you. If not, check which python-statemachine version is installed. It has to be 2.2 for now, not the latest 2.3
> 
> https://github.com/photobooth-app/photobooth-app/releases/tag/v3.1.4

I will test later this day.
Appreciate your work and your fantastic fast response. Wonderful!

harakaio, 2024-06-09:
> Updated version released. I think it should fix it for you. If not, check which python-statemachine version is installed. It has to be 2.2 for now, not the latest 2.3
> 
> https://github.com/photobooth-app/photobooth-app/releases/tag/v3.1.4

Works as expected. Wonderful, you did a great job!!

commit e34d1a5e (2024-06-09): pin python-statemachine to 2.2 as 2.3 causes issues currently. fixes #291