photobooth-app/photobooth-app #299: Wigglecam to create wigglegrams

mgineer85, 2024-06-16:
### What are wigglegrams?
https://www.reddit.com/r/wigglegrams/

You need a camera with at least two lenses to create wigglegrams
I created an experimental wigglegram camera: https://github.com/photobooth-app/wigglecam
Not sure yet, where to go from here and how to integrate with the photobooth or better keep it separate.

Would you like it?

### Find out more
[image]
[image]

-> find more information: https://github.com/photobooth-app/wigglecam

PixelAlarm, 2024-06-16:
Really nice.
Would it be even possible to integrate more than 2 cameras? I saw an Event-booth with like 15 cameras and strong 3d-effect.
Synchronizing gphoto2-cams will be tough, I think.

In my opinion, it would be nice to add to a Photobooth. Would be really Special and an USP.

mgineer85, 2025-10-25:
This made some progress recently. Unfortunately it's not a mobile camera any longer but shall be integrated in a bigger photobooth box. This way it's easier to tinker around with the hardware and can be shrunken later anyways once the software/components chosen stabilized and the concept has proven to work.

The basic idea is still the same but the tools changed to something more appropriate:
Pi Camera v3 -> Pi Zero 2 -> Ethernet Adapter -> Switch Port 1 \
Pi Camera v3 -> Pi Zero 2 -> Ethernet Adapter -> Switch Port 2  \ __--> Swich Port 5 -> Pi4/Pi5 hosting photobooth app
Pi Camera v3 -> Pi Zero 2 -> Ethernet Adapter -> Switch Port 3  /
Pi Camera v3 -> Pi Zero 2 -> Ethernet Adapter -> Switch Port 4 /

The app connects to the Pi Zeros via Network using pynng to transport messages. The messages are High resolution images, low resolution images (livestream) and trigger to capture.

## Todo

- [ ] Implement picamera2 synchronization (is available now natively in picamera2).
- [ ] Finish the mediaprocessing pipeline to consume 2...4 (or more) images from a camera array.
- [ ] Use HDR mode in picameras for better quality (resolution is sufficient for wigglegrams).
- [ ] Maybe add ansible scripts to maintain several devices easily.
- [ ] Add tests
  - [ ] for mediapipeline
  - [ ] for backend/pynng connector

## other refs

-> https://github.com/photobooth-app/photobooth-app/blob/main/src/photobooth/services/backends/wigglecam.py
-> https://github.com/photobooth-app/photobooth-app/blob/dev/src/photobooth/services/backends/wigglecam.py
-> https://github.com/photobooth-app/wigglecam

For a mobile camera, this one might be great, once sold: https://www.reddit.com/r/wigglegramProject/

closed 2025-11-30 by mgineer85