photobooth-app/photobooth-app #300: Doesn't work with the new NumPy 2.0.0 (released 16th of june)

WebfinityDesign, 2024-06-17:
**Describe the bug**
A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

**To Reproduce**
- Take a freshly installed SD card with Raspberry OS.
- Install Photobooth-app
- Start photbooth

**Expected behavior**
Photobooth will start

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit

**Additional context**
NumPy 2.0.0 was released yesterday. I'm going to try to downgrade NumPy but I have to find out how yet

mgineer85, 2024-06-18:
While using numpy 2 in the photobooth-app is working fine, this seems not to be true for every other dependency.
For now I pinned numpy to <2, a new version is released.

If you installed globally, please do not expect the system not to break if passing --break-system-packages to pip. I recommend pipx installation method.

closed 2024-06-18 by mgineer85

WebfinityDesign, 2024-06-18:
Thank you for your quick response!