photobooth-app/photobooth-app #304: Audio with the videos

WebfinityDesign, 2024-06-19:
First I want to give you my compliments for this great job!

**Is your feature request related to a problem? Please describe.**
It would be nice to have the possibility to also have sound with video.
That way you can have a video guestbook ( like an audio guestbook, but even better). It can be used for example to congratulate someone with his/her birthday or wish a newly married couple a happy marriage.

**Describe the solution you'd like**
Be able to grab audio with videos.

WebfinityDesign, 2024-06-21:
I see the label help-wanted. I cannot code. Is there another way I can help?

mgineer85, 2024-06-23:
Hey, thank you for your support.
I do not think that there is another way than to code.
It's to signal people from the community to help out as I have no plan currently to tackle this one as I have no need for this and no hardware to test on windows and linux to ensure platform independence.

But I think it would be a good addition, so I keep it here on the backlog.

PixelAlarm, 2024-06-30:
Hey,

a possible solution could be to take the clips in camera via gphoto2 and download the file instead of capturing the preview.
This way, the audio from the camera would be used.

Downside could be, that live view won‘t be available.

In my opinion this would be great, since it would improve the Video Quality a lot.

WebfinityDesign, 2024-07-03:
But that solution will only work with gphoto2, am I correct?
I was thinking about connecting a microphone to the PRI as I'm using an Arducam

PixelAlarm, 2024-07-04:
Yes, my suggestion would be gphoto2 only.

mgineer85, 2024-08-23:
To solve this, each backend would need it's own implementation to capture a video including sound:

- gphoto2: capture movie, hardware from camera would be used, not sure how livestream works at the same time then.
- picamera2: record_video avail in general, no idea what hardware is needed on the pi, not sure yet, how livestream works at the same time, but probably not an issue.
- virtualcamera: needs no sound actually.
- webcameras:
  - opencv2: no idea about if native support exists
  - v4l2py: no idea about if native support exists.
- digicamcontrol: not sure if we need digicamcontrol at all; I am not aware of any windows user and digicamcontrol seems not to be under active development any more.

WebfinityDesign, 2024-08-23:
Would this link be of any help? 

[](https://forums.raspberrypi.com/viewtopic.php?t=207697)

I do have a USB mike to try it on the USB port and I also have the Logitech C270 webcam (with built in mike) they are taking about