photobooth-app/photobooth-app #309: Choose layer order for collage pictures

PixelAlarm, 2024-06-25:
**Is your feature request related to a problem? Please describe.**
Im working on a collage setting, where it looks like pictures lay above each other.
For this effect to look nice, it would be nice to be able to select the order of the individual images.

**Describe the solution you'd like**
Have the option, wheather picture should be inserted to the top or background of other images.
(Now, images will be inserted on top all of the time)

**Describe alternatives you've considered**
Change the settings order of the individual pictures, so that every picture is in the right layer.
This way, the first picture (and in my experience main picture) is not in the first place.

**Additional context**
Actual implementation:
Image order: Top left, top right, bottom left, bottom right.
[image]

Possible improvement.
[image]

mgineer85, 2024-06-25:
I am afk right now but I think you can move the position of each image in the merge definition where you could delete one set also.
If you change the order, you also change the sequence to merge.

PixelAlarm, 2024-06-25:
That‘s the Alternative I referred to, but this will also change the image order.
I would like to preserve the image order, but change the Layer order.

PixelAlarm, 2024-06-26:
Perfect.
I think before/behind merge would bring a great benefit already - An explicit layer order with numbers would be the full solution.
Looking forward to this.