photobooth-app/photobooth-app #311: Issues with Camera Settings and Stream Recovery Post-Update

Peda1996, 2024-06-29:
### Environment
- **Version:** Updated from 1.x (dont know exact version anymore) to latest
- **Operating System:** Ubuntu 22.04
- **Camera:** Canon 2000d
- **Backend:** gphoto2

### Description
After updating to the latest version of the photobooth-app from version 1.x, I am encountering several for me critical issues related to camera functionality and stream recovery:

1. The camera no longer sets the configured ISO automatically for live streaming mode (after capture).
2. If the camera experiences a power outage, the stream does not recover; instead, the error screen remains displayed.

### Questions
- What changes were made in the latest version regarding camera settings and stream recovery functionalities?
- What is the purpose of `_on_configure_optimized_for_idle()` in the gphoto2 backend? Could the absence of this setting be affecting the auto ISO / Shutter functionality?
- Is there a possibility that a configuration change that I might have missed is causing these issues?

mgineer85, 2024-06-29:
Could you please also provide some logging file?

Peda1996, 2024-06-29:
I right now also notized that right before the new photo is taken, eg. when the timer starts, the ISO setting is changed to liveview, as can be seen in the log file. And after the photo was taken, the iso / shutter stays the same (eg. capture iso / shutter). So it seems that after the photo was taken the iso is not reset / eg. set back.

(found these things using the search phrase: "setting custom iso value")

```
2024-06-29 20:27:06,386 [    INFO] setting custom iso value: Auto (gphoto2.py:220)
2024-06-29 20:27:06,432 [    INFO] setting custom shutter speed: 1/20 (gphoto2.py:230)
2024-06-29 20:27:08,176 [    INFO] transition after: idle--(start)-->counting (processingservice.py:295)
2024-06-29 20:27:08,177 [    INFO] exit: counting from _counted (processingservice.py:306)
2024-06-29 20:27:08,177 [    INFO] enter: capture from _counted (processingservice.py:299)
2024-06-29 20:27:08,178 [    INFO] current capture (1/1, remaining 0) (processingservice.py:348)
2024-06-29 20:27:08,179 [   DEBUG] capture to mediaitem.path_original=PosixPath('media/original/20240629-202708-178661.jpg') (processingservice.py:360)
2024-06-29 20:27:08,179 [    INFO] WledService preset_shoot triggered (wledservice.py:142)
2024-06-29 20:27:08,179 [ WARNING] WLED module not connected, ignoring request (wledservice.py:154)
2024-06-29 20:27:08,213 [    INFO] disable viewfinder before capture (gphoto2.py:296)
2024-06-29 20:27:08,213 [   DEBUG] request to _wait_for_lores_image waiting until ongoing request_hires_still is finished (gphoto2.py:174)
2024-06-29 20:27:08,233 [   DEBUG] request to _wait_for_lores_image waiting until ongoing request_hires_still is finished (gphoto2.py:174)
2024-06-29 20:27:08,488 [   DEBUG] configure camera optimized for still capture (gphoto2.py:193)
2024-06-29 20:27:08,488 [    INFO] setting custom iso value: 100 (gphoto2.py:220)
```

For the reconnection of the camera i sadly didnt find any hints within the log.

Here is the full log file:
[log.txt](https://github.com/user-attachments/files/16041854/log.txt)

PixelAlarm, 2024-06-30:
I experience the same behaviour with ISO not being set after capture. When triggered again, the ISO is set shortly after pressing the button.

For me this was ok, because the darker live stream is some kind of idle screen.

When taking a collage, ISO is not changing with the second picture onwards, so this seems more Like a bug.

mgineer85, 2024-06-30:
Hey, gphoto2 issue to reconnect should be fixed in above commit.

mgineer85, 2024-06-30:
And above should improve behaviour for setting iso/shutter.

mgineer85, 2024-06-30:
I think this needs a bit more elaboration. For Picameras it is different than for gphoto2, I think:

countdown for still:
- picamera runs in **still-mode** (to avoid delay due to switch actually when the picture shall be taken)
- gphoto2 runs in **video-mode** (so preview is good)

Peda1996, 2024-07-01:
Thank you for the quick resolution to the previous issues. The described cases are now functioning correctly. However, I have identified another (in this case small) problem:
- When the application starts, the live view ISO is not set immediately. (eg. in my case the ISO of the camera is used)
- Additionally, after recovering the camera from a power outage, the ISO is also not set. (eg. in my case the ISO of the camera is used)

Could you please look into these issues? Thank you!

mgineer85, 2024-07-02:
Small above change should fix both. Would you test it?

Peda1996, 2024-07-02:
Thanks for the fix, they seems to be working flawlessly on my system :)

mgineer85, 2024-07-03:
OK, closing it here, will be included in next release later.
Thank you for testing, this is very helpful!

mgineer85, 2024-07-12:
next approach to finally fix this.
different backends work internally slightly different. picamera2 for example shall be set to HQ mode on start of countdown, to avoid delay when switching on the actual capture.
gphoto2 can apply new settings to the camera in no time, so its better to switch mode right before the capture instead the countdown start.

automated testing is running now, I think it should work now in every case.

PixelAlarm, 2024-07-12:
Looking forward to this.
Will it be released shortly?

Would like to test this, but can‘t update to newest commits, since I‘m using pipx Version.

Peda1996, 2024-07-12:
Hey, sorry to inform you, but I think the provided fix does not work. As after the selection screen (eg. If one would like to retake a photo eg for collages) the iso and shutter is still set to capture.

mgineer85, 2024-07-14:
Hm, in processing, when the countdown is started, the signalbackend_configure_optimized_for_hq_preview function is called, which should result in finally call _configure_optimized_for_idle_video for gphoto2 when lores stream is requested.

I will test again later using hardware again.

PixelAlarm, 2024-07-14:
I tested yestetday.

Indeed, between collage pictures the video now starts with still-params and adjusts to video-params after a second or so.

I can just tell from seeing Auto-ISO for Video vs. Specific ISO set for flash.

mgineer85, 2024-07-14:
I tested on my canon 1100d now and the exif data is with ISO100 while livestream is with ISO800 - as set in the config.
if the camera needs 1 sec to apply this setting, it might never work as intended and it might need to stick to the capture settings even during captures.

mgineer85, 2024-08-18:
Closing this for now. Feel free to reopen if still an issue...

commit 1942c309 (2024-06-30): fix #311
commit ccdc7d99 (2024-07-02): fix #311
commit fd376444 (2024-07-12): next approach to finally fix #311
different backends work internally slightly different. picamera2 for example shall be set to HQ mode on start of countdown, to avoid delay when switching on the actual capture.
gphoto2 can apply new settings to the camera in no time, so its better to switch mode right before the capture instead the countdown start.
closed 2024-08-18 by mgineer85