photobooth-app/photobooth-app #313: Link to documentation of dl.php does not exist

WebfinityDesign, 2024-07-03:
**Describe the bug**
Link to documentation of dl.php does not exist
This is the link on row 3 of dl.php

**To Reproduce**
Steps to reproduce the behavior:
1. Open dl.php with an editor
2. Click on https://photobooth-app.org/extras/shareservice/ (row 3)
4. See error 

Not Found
The requested URL was not found on this server.

**Expected behavior**
Open the documentation

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser brave, chrome, firefox

**Additional context**
I'm not sure if it it supposed to be this url:

https://photobooth-app.org/extras/sync/

I tried that one, configured it for Dropbox, but then on:

chmod u+x ~/photobooth-app/boothupload.sh

 I get the error:

File or directory does not exist

mgineer85, 2024-07-03:
The default is 
        default="https://photobooth-app.org/extras/shareservice-landing/",
        
        
and works. Maybe it was changed during testing?

WebfinityDesign, 2024-07-03:
I used this link:
[download dl.php](https://github.com/photobooth-app/photobooth-app/blob/main/extras/shareservice/dl.php)

on this page
https://photobooth-app.org/setup/shareservice/

If you download the dl.php from that link it seems to be a different one.

mgineer85, 2024-07-08:
I see, URL in dl.php is updated, thank you for the hint.
✨

closed 2024-07-08 by mgineer85