photobooth-app/photobooth-app #314: On download open the share screen on the phone

WebfinityDesign, 2024-07-07:
**Is your feature request related to a problem? Please describe.**
When I download a foto from a dropbox link, I am presented the share menu on my (Android). On this menu I'm able to send the photo to whatever app to my phone. I can save it, send it to Whatsapp (even to my own account), Telegram, etc.
That way you're very flexible on storing the photo.
I hope it makes sense what I'm talking about. I'm not sure it is something specific from Dropbox, I think it it something in Android because it is used a lot to select with app you want to use to open the file. 

**Describe the solution you'd like**
Open the app selector on the phone as default behavior

[image]

mgineer85, 2024-07-08:
When you downloaded an image to the phone, you should just be able to hit the share-button and the menu depicted in your screenshot will appear.    What exactly would you expect to be different?

WebfinityDesign, 2024-07-08:
When the photo is downloaded it shows up in the browser. It would be nice if the share menu would pop up automatically. Then it's easier for the guests to decide what to do with the photo. I think some people don't know how to save it when it shows up in the browser. Especially the older people.

Peda1996, 2024-07-11:
I like your idea, for this reason I created a PR, where users can directly click on "Download" or "Share" within the browser.

mgineer85, 2024-07-12:
Thank you @Peda1996 for the patch, it will be included in the next release!

closed 2024-07-12 by mgineer85