photobooth-app/photobooth-app #316: Add Hardware Acceleration to FFMPEG Command for Video Processing

Peda1996, 2024-07-09:
## Description
To improve the efficiency and reduce the processing time of our video generation pipeline, we should incorporate hardware acceleration into our FFMPEG commands. 

## Proposed Changes
### NVIDIA (NVENC)
- Add `-hwaccel cuda` to enable CUDA hardware acceleration.
- Use `-c:v h264_nvenc` to utilize NVIDIA's hardware-accelerated H264 encoder.
- Include `-preset fast` to balance encoding speed and output quality.

### Intel (QSV)
- Add `-hwaccel qsv` to enable Quick Sync Video hardware acceleration.
- Use `-c:v h264_qsv` to utilize Intel's hardware-accelerated H264 encoder.
- Include `-preset fast` for encoding speed and quality balance.

### AMD (VCE)
- Add `-hwaccel vdpau` to enable VDPAU hardware acceleration.
- Use `-c:v h264_amf` to utilize AMD's hardware-accelerated H264 encoder.
- Include `-preset fast` for encoding speed and quality balance.

### Code Location
The current FFMPEG command is located in the `video_pipelinestages.py` file. Specifically, it can be found in the following lines:
[video_pipelinestages.py#L42-L49](https://github.com/photobooth-app/photobooth-app/blob/e94aa8fe1cab98c0044ea4089f04466e8d69e582/photobooth/services/mediaprocessing/video_pipelinestages.py#L42-L49)

The current command is:

```python
command_video_output = [ 
    "-filter_complex", 
    "[0:v]reverse[r];[0:v][r]concat=n=2:v=1[outv]", 
    "-map", 
    "[outv]", 
    "-movflags", 
    "+faststart", 
]

```
### Expected Benefits
- Reduced video processing time.

mgineer85, 2024-07-09:
Do you have any numbers about the actual speed up?

I considered to implement some kind of hardware acceleration but hesitated since the implementations can produce different quality, could fail if wrong accelerator is chosen and can have different parameter as well as limitations (example regarding max width/height of an video).

Still I think it could be good if performance increases.

Peda1996, 2024-07-09:
Well, for now I don't have actual speed up numbers, maybe tomorrow I might find time to fiddle around with my booth setup and add the nvidia hardware acceleration parameters. 

Maybe an "easy" solution to the problem would be to add "expert" ffmpeg parameters, where users can add hardware acceleration parameters on their own, and fiddle therefore with their own setup? And maybe then add Infos in the docs for what system what params to use. With this also quality parameter etc. could be set by the user.

A more advanced solution would incooperate checking (when starting up) what hardware acceleration is present in the system, and then only apply the selected hardware acceleration if the system can handle the selected acceleration.

Regarding width / height I don't realy have any idea either to solve this..

Peda1996, 2024-07-11:
I tried to get hardware acceleration working, but wasn't able to install the h264_qsv codec in Ubuntu.

Botts85, 2024-08-30:
> Do you have any numbers about the actual speed up?

I saw a significant performance uplift, but I'm using a relatively light host for photobooth-app on Windows. It's an Intel M5-6Y57 with only two cores.

With Intel QSV acceleration on the encode a 5 second clip processes in about 7.5 seconds.
Without QSV acceleration it took 14 seconds to process.