photobooth-app/photobooth-app #317: [ WARNING] error connecting to service: HTTPSConnectionPool foe qr sharing

Florian-Costel, 2024-07-10:
Hi!
I am having trouble to setup the sharing by QR code. I follow the setup instructions and configure dl.php.
After scanning the QR code, on the webpage I receive the same error as the php-error.log:

 runtime error: photobooth did not upload the requested image within time :( no internet? service disabled? {"action":"download","id":"a74af332f9ee63c76592d0e608e1c7e3"}

On the photobooth app I have this error:
2024-07-10 12:24:59,461 [ WARNING] error connecting to service: HTTPSConnectionPool(host='foto.xxx.com', port=443): Read timed out. (read timeout=8) (shareservice.py:85)

Do you have any suggestion for this issue?

Thanks,
Florian

mgineer85, 2024-07-10:
Another user is having the same issue currently in the discussions. I think it is related to the php host that is not reachable by the apps service.

WebfinityDesign, 2024-07-10:
Hi Florian,

I don't have a solution for you, but I do have the same problem. It seems to me that the place the dl.php is hosted has something to whit it. Can I ask you where your dl.php is hosted? Is it Apache or Nginx? Is it a normal hosting for websites or a VPS?
I tried different places with different results, though not one yet that is fully functional.

mgineer85, 2024-07-10:
another reference: https://github.com/photobooth-app/photobooth-app/discussions/312#discussioncomment-10009391

mgineer85, 2024-07-10:
I just tried again on my cheap shared webhosting, that worked well.

Please delete the sqlite database, I don't know yet why, but there is something going on and it tries to reupload a failed media every time again.

Is the php max upload limit set too low maybe?
Do you have by accident maybe two app instances running at the same time pointing to the same dl script? That is very likely to fail as it is not designed for this use case.

WebfinityDesign, 2024-07-10:
I just deleted the sqlite database again. I did that before when I saw it kept trying a certain media file over and over again.
It doesn't help. It is still trying to upload one and the same file over and over again.

I cannot find the php max upload limit. I attached a screenshot of my php settings.

[image]

Florian-Costel, 2024-07-10:
Hi all!
I follow your advice and moved the dl.php host and now is working fine. Initially it was hosted on a simple vps with nginx and PHP Version 7.2.24. Now the dl.php is on a normal Cpanel space with apache2 web server and PHP Version 7.4.33.
Probably the issue is related with nginx or some missing php package.

Thank you!
Florian

WebfinityDesign, 2024-07-10:
I am running on PHP 8.1. I will try 7.4 and see if there is any difference

mgineer85, 2024-07-10:
The automated tests run on 8.1 also. So maybe it’s still something different

Peda1996, 2024-07-11:
Hey, I think I had the same problems with my original nginx configuration. https://github.com/photobooth-app/photobooth-app/issues/140 

Maybe this will help

WebfinityDesign, 2024-07-11:
Hey, thanks for your reply. I've been looking at your post before. Maybe that could be the solution, but I don't know how to implement it in my situation. Can you elaborate a little what is causing the problem?

Peda1996, 2024-07-11:
Considering:

```
    location / {
        try_files $uri $uri/ /index.php$is_args$args;
    }

	location ~ \.php$ {
    	include snippets/fastcgi-php.conf;
    	fastcgi_pass unix:/run/php/php8.2-fpm.sock;
    	fastcgi_read_timeout 500;  # Set this to a value that suits your needs
        fastcgi_buffering off;
	}

    location ~ /\.ht {
        deny all;
    }
```
The main fix is due to` fastcgi_buffering off`, which ensures ensures that responses are sent to the client as they are generated by PHP-FPM, so no buffering stage is between. 

And further the `fastcgi_read_timeout 500` which increases the amount of time Nginx will wait for a response from the FastCGI server (PHP-FPM) before timing out. (I thought this might help with slow internet connections). Additionally I set the `client_max_body_size 50M;` to allow bigger filesizes (eg. for videos and such).

Peda1996, 2024-07-11:
```
// disable output_buffering to work with long requests
while (ob_get_level()) {
    ob_end_flush();
}

// Turn off output buffering
ini_set('output_buffering', 'Off');

// Disable output compression
ini_set('zlib.output_compression', 'Off');

// Enable implicit flush
ini_set('implicit_flush', 'On');
ob_implicit_flush(true);
```

This part within the dl.php can also fix the issue. I think I will create a PR for this. Can you try to add this in the beginning the line after <?php ?

Peda1996, 2024-07-11:
Can you try to add 

```
// Setting this header instructs Nginx to disable fastcgi_buffering and disable
// gzip for this request.
header('X-Accel-Buffering: no');
```

after line after <?php

This also fixes the problem in my case, but im using nginx..

WebfinityDesign, 2024-07-11:
Thanks,
I have added both pieces of code, but I still get the response:

`[    INFO] request timed out, error occured or shutdown requested (shareservice.py:179)
`

Peda1996, 2024-07-11:
Can you try this line? Sry that I cannot test it with apache, as I don't
use it 😅.

ini_set('output_buffering', 'Off');

mgineer85, 2024-07-11:
> Can you try this line? Sry that I cannot test it with apache, as I don't use it 😅. ini_set('output_buffering', 'Off'); Th3o ***@***.***> schrieb am Do., 11. Juli 2024, 14:56:

This should not be neccesary because on long running job the script flushes anyways:
https://github.com/photobooth-app/photobooth-app/blob/e94aa8fe1cab98c0044ea4089f04466e8d69e582/extras/shareservice/dl.php#L189

mgineer85, 2024-07-11:
I pushed an improved dl.php: https://github.com/photobooth-app/photobooth-app/commit/230cd9fe54059b158a0baea51f31c2f5e474965c

Maybe someone can test and confirm it works better for nginx?
I don't think it will change anything for the apache.

Peda1996, 2024-07-11:
I can test it with nginx tomorrow. I assume it should work with nginx because when I added `header('X-Accel-Buffering: no');`, it worked flawlessly.

WebfinityDesign, 2024-07-11:
> Can you try this line? Sry that I cannot test it with apache, as I don't use it 😅. ini_set('output_buffering', 'Off');

No problem. I'm not a coder, so I'm glad I can help in another way by testing things.

I tried this line but it didn't make any difference. There is one strange thing: because the line didn't work I removed it from the dl.php again. After I started the PB app again it worked for one time! After that I got the green runtime errors again. Seems to me it has something to do with a timing problem.

WebfinityDesign, 2024-07-11:
> I pushed an improved dl.php: [230cd9f](https://github.com/photobooth-app/photobooth-app/commit/230cd9fe54059b158a0baea51f31c2f5e474965c)
> 
> Maybe someone can test and confirm it works better for nginx? I don't think it will change anything for the apache.

I tested it on my Apache webhosting and it didn't make any difference (as expected)

Peda1996, 2024-07-11:
Another thing that you could try would be to create a .htaccess file where
you add

# Disable output buffering (most important)

php_flag output_buffering Off

# Increase script execution time (in seconds)

php_value max_execution_time 300

# Increase memory limit (in MB)

php_value memory_limit 256M

# Increase the maximum upload file size (in MB)

php_value upload_max_filesize 50M

# Increase the maximum POST size (in MB)

php_value post_max_size 50M

# Allow long-running requests

<IfModule mod_reqtimeout.c>

    RequestReadTimeout body=120,minrate=1

</IfModule>

# Enable output buffering to flush response data to the client as soon as
possible

<IfModule mod_headers.c>

    Header set Content-Encoding none

</IfModule>

# Enable keep-alive connections

<IfModule mod_headers.c>

    Header set Connection keep-alive

</IfModule>

# Increase the default socket read timeout

<IfModule mod_proxy.c>

    ProxyTimeout 600

</IfModule>

# Set Apache timeout for long requests

Timeout 600

# Ensure Apache doesn't terminate PHP scripts prematurely

<IfModule mod_fcgid.c>

    FcgidIOTimeout 600

</IfModule>

# For Nginx users behind reverse proxy

<IfModule mod_rpaf.c>

    RPAFenable On

    RPAFsethostname On

    RPAFproxy_ips 127.0.0.1

</IfModule>

This answer was generated by chatgpt, but maybe it will work 😅.

WebfinityDesign, 2024-07-11:
I tried it, but it gives this response:
response from dl.php script: <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>500 Internal Server Error</title>
</head><body>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error or
misconfiguration and was unable to complete
your request.</p>
<p>Please contact the server administrator at 
 [no address given] to inform them of the time this error occurred,
 and the actions you performed just before this error.</p>
<p>More information about this error may be available
in the server error log.</p>
<hr>
<address>Apache/2.4.59 (Debian) Server at xxxxxxxx.com Port 80</address>
</body></html>

mgineer85, 2024-07-12:
> The access from outside to my home server on my home network is running through a Cloudflare tunnel. Could that be an issue?
> The photobooth itself is through the router directly connected to the internet as far as I know.

Did you in the meantime remove the cloudflare from the equation?
Just upload the dl.php to any shared hoster in the internet, not in your local network.

WebfinityDesign, 2024-07-12:
Besides the dl.php on my home server I uploaded it to two different web hosting servers and one VPS.
The only time I got it working is when all the communication is running through my local network. 
Which web hosting service do you use yourself? For me that could be a temporary solution until we have pinpointed the real cause of this.

mgineer85, 2024-07-12:
I use all-inkl.com

WebfinityDesign, 2024-07-12:
I registered an account there now. See how that goes. I have to wait 1 or 2 days before the domainname is active

mgineer85, 2024-07-12:
Not via domain, but every account there usually has some temporary dns name that could be used until full domain is avail. 
I think they also got regular test accounts.

WebfinityDesign, 2024-07-12:
It is already active! Download is working now with the same configuration that is having trouble on other hosting servers. So it must be something in the differences of the webhosting configurations

mgineer85, 2024-07-12:
Great! If you can find any difference and solution, I would love to add it to the docs!

WebfinityDesign, 2024-07-12:
I would be happy to search for the differences, but to be honest I don't know where to look other than the phpinfo

WebfinityDesign, 2024-07-12:
This is an errorlog from a "not working with dl.php" server

[image]

mgineer85, 2024-07-13:
I think all servers I used have the php included as module not fastcgi.
I think you need to track down why this issue is occurring: https://stackoverflow.com/questions/18680949/high-cpu-mod-fcgid-cant-apply-process-slot/19215155#19215155

mgineer85, 2024-07-14:
I close this as I think it is about the webhosting configuration.
Whenever you find some hints where things go wrong for the fastcgi type of hosting (or whatever else it might be), feel free to reopen.
Thank you for your testing!

closed 2024-07-14 by mgineer85

WebfinityDesign, 2025-01-13:
I'm trying to use dl.php on another host. When scanning the QR-code I get:

504 Gatweway Time-out
nginx

In the PHP error log on the host I get these errors:

[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:41 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:42 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:42 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:42 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:42 UTC] PHP Warning:  SQLite3::querySingle(): Unable to execute statement: database is locked in /home/xxxxxxxx.xxx/httpdocs/dl.php on line 224
[13-Jan-2025 12:59:57 UTC] Runtime error: photobooth did not upload the requested image within time :( no internet? service disabled?
[13-Jan-2025 12:59:57 UTC] {"action":"download","id":"b541eb6955d98e97b2b29fd308e32b52"}
[13-Jan-2025 12:59:57 UTC] []
[13-Jan-2025 12:59:57 UTC] []

When I ask ChatGPT I get this as possible causes:
- Too many simultaneous connections
- Long database edits
- Connections not properly closed

Is this helpful?

mgineer85, 2025-01-13:
Hm, no PHP expert actually.

Would it improve if you modify the script to use the WAL mode?
For example like this: 
```
    // db connection setup
    $db = new SQLite3($DB_FILENAME);
    $db->busyTimeout(2000);
    // WAL mode has better control over concurrency.
    // Source: https://www.sqlite.org/wal.html
    $db->exec('PRAGMA journal_mode = wal;');
```

WebfinityDesign, 2025-01-13:
It seems like something changes. The error about the database locked seems to be gone.

The only PHP error that is left is:

[13-Jan-2025 17:06:57 UTC] Runtime error: photobooth did not upload the requested image within time :( no internet? service disabled?
[13-Jan-2025 17:06:57 UTC] {"action":"download","id":"12641b3809bb95e758cd62506d8ee6b0"}
[13-Jan-2025 17:06:57 UTC] []
[13-Jan-2025 17:06:57 UTC] []
[13-Jan-2025 17:07:06 UTC] Runtime error: photobooth did not upload the requested image within time :( no internet? service disabled?
[13-Jan-2025 17:07:06 UTC] {"action":"download","id":"12641b3809bb95e758cd62506d8ee6b0"}
[13-Jan-2025 17:07:06 UTC] []
[13-Jan-2025 17:07:06 UTC] []

================================
In the access log I see a lot of 
"GET /dl.php?action=upload_queue HTTP/1.0
Are that retries?

=====================================

In the error log of the webserver I see a number of connection resets:

[Mon Jan 13 17:17:08.772438 2025] [fcgid:warn] [pid 3744770:tid 3744832] (104)Connection reset by peer: [client xxx.xxx.xxx.97:0] mod_fcgid: error reading data from FastCGI server
[Mon Jan 13 17:17:08.772597 2025] [fcgid:warn] [pid 3744770:tid 3744832] (104)Connection reset by peer: [client xxx.xxx.xxx.97:0] mod_fcgid: ap_pass_brigade failed in handle_request_ipc function

============================================

[error log.txt](https://github.com/user-attachments/files/18400096/error.log.txt)
[access log.txt](https://github.com/user-attachments/files/18400097/access.log.txt)

mgineer85, 2025-01-13:
It looks like the server is overloaded maybe?

there are many 503 errors. https://developer.mozilla.org/de/docs/Web/HTTP/Status/503

did the server change recently? any updates?

> ================================
In the access log I see a lot of
"GET /dl.php?action=upload_queue HTTP/1.0
Are that retries?

No, this is not an actual try to upload an image but to establish a long running connection from the photobox to the server. It's using this connection to receive a notification when a user scans a qr code.

WebfinityDesign, 2025-01-13:
According to ChatGPT is it most likely that the webserver doesn't accept outgoing connections. That seems to be a problem with a lot of webhostings

mgineer85, 2025-01-13:
Webserver serve incoming connections, there is no outgoing connection from the Webserver.

WebfinityDesign, 2025-01-14:
To be honest I don't know enough about coding to check if what ChatGPT says is correct.
I'm trying to find the cause of the dl.php not working on a lot of webhostings. I think I tried something like 7 or 8 hostings and the only one I can get it working is on all-inkl.de.
It would be nice if you can upload it to just any webhost and it just works.

As far I can see it is not caused by the regular PHP settings like timeouts etc. I compared them with all-inkl.de and with more or less the same settings I can't get it to work

mgineer85, 2025-01-14:
I think about replacing the solution by a ftp sync option. That would have just the ftp server dependency and might be more simple.

but it would mean all files need immediate upload instead on request only.

WebfinityDesign, 2025-01-15:
Ok, let me know if I can test something for you

mgineer85, 2025-01-15:
Maybe other way around first, could you list providers you used that did not work? 
Maybe I can have a test account to see what goes wrong.

It is probably earlier tested than a new implementation is avail