photobooth-app/photobooth-app #322: Share service not working when internet is not available on startup

Peda1996, 2024-07-15:
The QrShareService in the photobooth application does not function correctly when the application is started without an internet connection. The service should periodically check for an available internet connection and enable the share service once a connection is detected.

## Steps to Reproduce:
1.  Ensure that there is no internet connection.
2.  Start the photobooth application.
3.  Attempt to use the share service.

## Code Snippet:
Error happens in this section. It is checked on startup if the endpoint returns code 200. But if no internet connection is avaliable to beginn with, its not possible and non 200 is returned, and therefore the whole service stops, despite propably connecting to the Internet later on.
https://github.com/photobooth-app/photobooth-app/blob/b24bf5673c70f14900d66314d244447ae66ba9dd/photobooth/services/qrshareservice.py#L28-L47

## Expected Behavior:
The application should periodically check for an internet connection and enable the share service once a connection is available.

mgineer85, 2024-07-15:
I just removed the initial checks for a valid endpoint. It's probing now every 5 seconds - should be acceptable compromise between checks and not overly often repeat.

closed 2024-07-15 by mgineer85

Peda1996, 2024-07-15:
Perfect, thanks for this fix :)

commit 51ea895c (2024-07-15): fix #322