photobooth-app/photobooth-app #327: Add direct printing ability

Apfelraeuber, 2024-07-20:
**Is your feature request related to a problem? Please describe.**
Many photobooth systems are able to print directly after an image has been captured. The idea is rather simple, after an image or a series of images (in case of an collage) have been taken, the image gets directly send as a print job to the printer. Especially if the trigger is realized with a red button, I see the tendence of users not to make use of the touchscreen. So most of them hesitate or do not recognize that it's actually possible to print the captured image.  

**Describe the solution you'd like**
An additional setting for images and collages to use "direct print". If this setting is enabled, captured images or collages get directly printed without user interaction. 

**Describe alternatives you've considered**
The usual printing button within the software. Also a standalone gallery touchscreen. But people hesitate to actually print. 

I would love to contribute, but had little time to get in touch with the code base. My first thought was adding it to photobooth/services/config/groups/actions.py. If i have some spare time I will look into this in more detail.

mgineer85, 2024-07-21:
In v4 there is not just one print command but many configurable.
So following would need to be considered for implementation 
- ability to choose a print command among the configured set
- Option to ignore videos/gifs for automatic printing (despite other share commands could work for videos also)

Apfelraeuber, 2024-07-29:
Jep, I've seen that. I thought of adding it as an individual option to the Image and collage action settings.