photobooth-app/photobooth-app #329: No module named 'photobooth'

j79, 2024-07-24:
Hello,

I made a fresh install on an RPI5 with a webcam.
When pushing this command 
`"python -c "from photobooth.services.backends.webcamv4l import *; print(available_camera_indexes())""`
I get this result
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'photobooth'

```
I use a logitech C930. I've tested it successfully with VLC

I'm sure I'm not far but I do not find the solution.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser Opera
 - Version 112.0.5197.30

mgineer85, 2024-07-24:
Hey, what installation method was used? Probably on a fresh system you just need to reboot

j79, 2024-07-24:
I used METHOD C

Tried to reboot without succes so I went into the config and have tested 0 then 1 then 0 again for WebcamCV2 and the app is running quite well.

It's so not a blocker but I thought it could be interesting for you to know that.

mgineer85, 2024-07-25:
Thank you, I will check it

mgineer85, 2024-07-25:
The command itself is tested and works properly.

Maybe it is because Method C is actually not recommended any more on newer systems because it interferes with system packages.

I am closing this as there is no actual issue.

Have fun with your booth :)

closed 2024-07-25 by mgineer85