photobooth-app/photobooth-app #334: QR-code share does not work on Chrome for Android, Firefox for Android and Brave for Android.

WebfinityDesign, 2024-08-12:
**Describe the bug**
QR-code share does not work on Chrome for Android, Firefox for Android and Brave for Android.

**To Reproduce**
1. open the URL of the QR-code in one of the previous mentioned browsers
2. The picture will show with a download and a share button
3. Press share
4. The browser responds with "Web share is not supported in your browser"

**Expected behavior**
After pressing the share see a menu with app you can use to share the photo

**Screenshots**
[image]
**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser: chrome, firefox, brave

**Additional context**
When pressing the download button the download succeeds, but it would be nice to have a confirmation dialog. Something like "download completed" or "succesfully downloaded"

Peda1996, 2024-08-12:
> :
> Hey @Peda1996 <https://github.com/Peda1996> , you got an idea what could
> be wrong?
> I think maybe the benefit added by the share button is limited compared to
> the additional issues and maintenance.

Hey, after a quick google search I found out that Web Share API is only
available in a *secure context* (
https://stackoverflow.com/questions/69788720/my-web-share-api-is-supported-by-chrome-in-localhost-but-not-supported-once-i-ho).
Meaning that webshare is only available with the usage of *https://.* To
enable this, maybe use *letsencrypt *or something similar.

Peda1996, 2024-08-12:
To provide a "fix" for this, we might also add a check whether the url
scheme is https or not, and only add the share button when https is used.

if ($url['scheme'] == 'https') {
       // is https;
   }

For clarification we should then also add a small documentation
comment at the beginning of the script, that tells users that the
share functionality is only available via https..

Peda1996, 2024-08-12:
Please test it, as I don't have my photobooth setup here with me this week.

WebfinityDesign, 2024-08-12:
I think I will make a dedicated PHP webserver on an ESP32. Then see if I can host the dl.php on that server.  It can be connected to your home network. If we can get that working we can make a working image for the ESP32. That way you don't have to fiddle with hosters and see if they work or not

Peda1996, 2024-08-12:
If you want to do it that route, you could just setup your own nginx, add DuckDNS and LetsEncrypt. Similar like everyone has to do it on their own VPS 😅

Working Example nginx config:
```
server {
    server_name example.com;
    client_max_body_size 50M;
    root /var/www/html/shareservice;
    index index.php;

    location / {
        try_files $uri $uri/ /index.php$is_args$args;
    }

	location ~ \.php$ {
    	include snippets/fastcgi-php.conf;
    	fastcgi_pass unix:/run/php/php8.2-fpm.sock;
    	fastcgi_read_timeout 500;  # Adjust this value as needed
        #fastcgi_buffering off;
	}

    location ~ /\.ht {
        deny all;
    }

    error_log /var/log/nginx/shareservice.error.log;
    access_log /var/log/nginx/shareservice.access.log;

    listen [::]:443 ssl;
    listen 443 ssl;

    ssl_certificate /etc/letsencrypt/live/example.com/fullchain.pem;
    ssl_certificate_key /etc/letsencrypt/live/example.com/privkey.pem;
    include /etc/letsencrypt/options-ssl-nginx.conf;
    ssl_dhparam /etc/letsencrypt/ssl-dhparams.pem;
}

server {
    if ($host = example.com) {
        return 301 https://$host$request_uri;
    }

    listen 80;
    listen [::]:80;

    server_name example.com;
    return 404;
}

```

mgineer85, 2024-08-13:
> I think I will make a dedicated PHP webserver on an ESP32. Then see if I can host the dl.php on that server. Also include DuckDNS and Let's encrypt. It can be connected to your home network. Maybe you can even add it to your photobooth if it has his own mobile internet connection. If we can get that working we can make a working standard image for the ESP32. That way you don't have to fiddle with hosters and see if they work or not

Don’t do that.
The reason the qr service exists is that it works around the limitation of mobile connections that don’t allow to open ports.
If you want to have a local solution, just sharecthe mediations in the local network via QR code.

WebfinityDesign, 2024-08-13:
Sorry, I don't understand exactly what you mean. Do you mean a direct connection from the phone to the ESP32 server?
My goal is to have a simple ESP32 server connected to my home internet connection in a way that I can connect to it from the internet replacing the internet hosting provider to host the dl.php. So being my own (mini) hosting provider.
Maybe I'm saying something stupid here, so please correct me in that case.

mgineer85, 2024-08-13:
I think just a random shared Hoster you put the dl.php to is the most hassle free way to go.
I don’t think you need to add another device to the equation.

If you only want to share locally you can setup QR code to download directly from the raspberry pi.

If you want external you can use dl.php or create your own sync solution of the data for. Then you point to the external files with the QR code.

WebfinityDesign, 2024-08-14:
I thought maybe it could be a solution for the hassle with the hosts that are not working with dl.php. Of the 5 hosts / VPSes I tried for me the only one working is the one one you use.

Peda1996, 2024-08-14:
I still think that this issue can be resolved using the correct .htaccess
file or additional php ini configs within the script:

https://stackoverflow.com/questions/16954468/output-buffering-off-at-run-time-in-php
https://stackoverflow.com/questions/8882383/how-to-disable-output-buffering-in-php

Maybe also setting these vars could help?
// Disable output buffering to ensure immediate output
ini_set('output_buffering', 'off');
ini_set('zlib.output_compression', 'off');

But, as I don't use apache, somebody else has to try these things..

WebfinityDesign, 2024-08-14:
I don't have the time today to try that out. I'm "photobooting" my sons 18th birthday :)
But I will try it out later

mgineer85, 2024-09-14:
merged the code, so the original issue is solved. thx @Peda1996

closed 2024-09-14 by mgineer85