photobooth-app/photobooth-app #337: Change the backend's first launch behavior to generate a fully populated default config file before attaching to the listening port

skysyk, 2024-08-12:
**Is your feature request related to a problem? Please describe.**
It seems like the app doesn't generate a full `config.json` file until a "successful" first launch, which includes the server attaching to the listening port. This becomes inconvenient when the default listening port is already in use since the launch will stall at the point and no config file will be populated. The user would need to manually create the file with the appropriate entries to change the listening port / interface.

**Describe the solution you'd like**
Generate the default config as early in the app launch process as possible so new users can easily modify critical parameters as needed.

mgineer85, 2024-08-18:
I'd propose to remove the settings and instead add it as a parameter during startup in the future.   creating the webserver should not actually be handled by the app itself.

mgineer85, 2025-01-14:
In v5 the program can be started differently.

The port/host-ip are not coded in the config any more but can be passed as an argument when the photobooth is started.
Like so:
`photobooth --host 0.0.0.0 --port 1234`

I close this issue as it will address the original issue also.

closed 2025-01-14 by mgineer85