photobooth-app/photobooth-app #339: USB Copy Notifications

Peda1996, 2024-08-12:
I would like to request a feature that provides notifications when data is copied automatically using the USB Background Copy functionality. 

## Expected Behavior
When the USB Background Copy functionality is triggered:
- A notification should be displayed to the user.
- The notification should include information such as the current progress, remaining items, and possibly an estimated time for completion.
- Additionally, an alert or notification should be shown if an error occurs during the copying process.

## Example Implementation
Below is an example of how this could be implemented using the existing notification system:

```python
self._sse_service.dispatch_event(
    SseEventFrontendNotification(
        color="info",
        message=f"Copying files to USB: {files_copied}/{total_files} files copied",
        caption="USB Copy Progress",
    )
)
```

In the case of an error, such as exceeding a quota or encountering a disk error:
```python
self._sse_service.dispatch_event(
    SseEventFrontendNotification(
        color="negative",
        message=f"Error during USB copy: {error_message}",
        caption="USB Copy Error",
    )
)
```

## Benefits
- Improves user awareness of background processes.
- Provides real-time feedback on the status of the USB copy operation.
- Enhances user experience by reducing uncertainty during data transfers.

mgineer85, 2024-08-13:
The screen would be flooded by the messages quite quickly.
Quasar supports updated messages. Not did have a look how it works, but seems to be useful here.

WebfinityDesign, 2024-08-14:
I just tested it. I like it and think it is quite useful. The screen is flooded by messages indeed, but if there is no other way to do it that is ok by me. Maybe a progress-bar would be better, but I don't know if that is an easy thing to do.
There is one thing would like to change, but I think that is some adjustment in the OS: when I plug in the USB stick I get a popup that a removable media is inserted. In case we have this notification the popup is not necessary. Btw. I also got that popup with the original filetransferservice.py.

[image]

Peda1996, 2024-08-14:
Oh, that looks horrible 🫣.

The easiest way of mitigating the problem would be to just periodically
show the remaining time. Eg every 5 seconds or so, then we don't need to
modify the frontend and the frontend will not be flooded with
notifications. But it's a kinda hacky, solution 😅.

WebfinityDesign, 2024-08-14:
That is just the popup screen I get when I insert the USB stick :))

The progress bar Ik was talking about is here (the previous link didn't work):

[[https://quasar.dev/vue-components/linear-progress/]([https://quasar.dev/vue-components/linear-progress/](https://github.com/photobooth-app/photobooth-app/issues/url)](url)

mgineer85, 2024-08-16:
Maybe there could be a more universal implementation displaying
- modal progress bar and some text or
- notification progress bar and some text.

First I could think of could replace the waiting screen during the capture process when the image pipelines are applied. Or in general: If something is going on the user should have to wait for it's outcome.
Second I think could be useful to signal some background tasks: Example ongoing print until block time ended, copy files, ... 

First would be an overlay probably, second a notification.   If backend and frontend implemented well, the notifications can be made updateable somehow...

dtimes, 2025-01-31:
maybe adding support for a small additional I2C display like the SSD1306/SSH1106 oleds might be a nice option to display state of the box and running tasks without interfering with the normal operation of the box , could be easily integrated with the case (side or backside) without drawing too  much attention and it´s easy to connect and quite cheap

mgineer85, 2025-01-31:
I am about to introduce a plugin-system in the next version.
Still trying to figure out, how it could work, but it could be used for extensions of all kind then.

mgineer85, 2025-06-03:
I will not actively implement this right now. Maybe it happens one day as many items from my (previously internal) list have happened.

I added an item to the project list to keep it in mind but closing this one now.

If anybody is willing to add this feature, the synchronizer plugin could probably extended in a way to report stats to the informationservice. The informationservice would send cyclic updates to the frontend and there it could be displayed in the admin dashboard for example.

-> https://github.com/orgs/photobooth-app/projects/2/views/1?pane=issue&itemId=113645442