photobooth-app/photobooth-app #345: No print button

Louloubiwan, 2024-08-23:
**Describe the bug**
[Uploading IMG_20240823_182957.jpg …]()

Hello, so there is no print button when photo taken even with the Show print true. If you know how to fix it, thank you !
 [IMG_20240823_183321.jpg](https://github.com/user-attachments/assets/eb141c3b-7b39-4f98-bf80-1b53c6a83255)

**To Reproduce**
Take a photo and try to print it

 - Raspberry Pi Bookworm 64bit
 - Chromium
 - Version : the last one

Striker62, 2024-08-26:
Hi, you can check in the share section of the config 

Number Direct Access Buttons
Number of buttons directly accessible in the gallery. Remaining items are available in the more-menu.

Does this fit with the no of buttons ?
If Yes it cloud be a browser cache problem.

mgineer85, 2024-08-26:
Yes could be cache or after update there is no new default share option configured. Go to the share actions and ensure there is one.

Also ensure the global switch share-enable is active in that section.

closed 2024-08-27 by mgineer85