photobooth-app/photobooth-app #347: Custom share action inputs

larsl-net, 2024-08-25:
**Is your feature request related to a problem? Please describe.**
If a group of people makes a photo and wants multiple prints, you need to click multiple times on the print action.

**Describe the solution you'd like**
With a custom input definition similar to the `Merge Definition` for `collages` you could configure an input dialog which opens after the first click on the share action button. In this, the user needs to enter the custom inputs if no default is set (number of prints, email, etc.).

These custom inputs need to be available as placeholders in the share command that the admin can build custom workflows with the user data.

**Describe alternatives you've considered**
Dedicated option to select number of prints.

**Additional context**
With this feature it would also be possible to add other more complex sharing options like send the pictures per email to the users.

mgineer85, 2024-08-26:
While it would be cool to add it’s a bit effort to do it right. Target main hardware would probably be a booth with touchscreen.

 I am not really up to date about the raspberry onscreen displays but I found them to work not very reliable and easy to use. That means we have to find a way implementing a keyboard on our own which makes it even harder to do good.

As an intermediate solution I could think of you create just 3 share commands with one, two and three copies. That should solve your main problem I think?

larsl-net, 2024-08-26:
Yes my current problem can be solved with multiple share actions.

In dslrbooth they have implemented a simple [+/- button solution](https://dslrbooth.com/blog/dslrbooth-for-windows-v5-32) for int values.

But for complex strings, like a email address, a good onscreen keyboard is required. I use the maliit on my work Lenovo Thinkbook with Fedora 40 and Plasma 6.1, which works good for the most applications, but for example in brave it doesn't open.

mgineer85, 2024-08-26:
Also I would guess guests could get creative on breaking out of the app using a full keyboard 😅

mgineer85, 2025-02-16:
Fiddeling around with this...

Currently there would be 2 types of input fields, a number with +- buttons and a normal input field.
[image]

The input fields could be configured as follows currently in the very basic version:
[image]

Working on this in a separate branch:
https://github.com/photobooth-app/photobooth-app/tree/shareInputApi

If a keyboard is connected, you could also require the user to input the mailadress or an phone number and the information is passed to the share command for further processing...

mgineer85, 2025-02-18:
Going to merge this now and have it in v6 soon: 

https://github.com/user-attachments/assets/079b5194-8cd8-453b-ac89-afd8273be900

all elements in the dialog field are configurable. the number input has validation and allows to be used without keyboard by clicking + and - buttons.
The mail address (or whatever else the command expects for execution) needs a keyboard of course.

mgineer85, 2025-02-18:
When there are any issues later with the implementation, please open a new issue :)

closed 2025-02-18 by mgineer85

PixelAlarm, 2025-03-07:
Hi, I really like this new feature. Congrats.

I was thinking about implementing a share function via mail, but there is no possibility to input a mail address right now, is it?

Sending mails via CLI is pretty easy and would allow to be used as simple command, but it needs the dynamic input of the recipients mail address.

mgineer85, 2025-03-07:
Yes, should be possible now.
But you need to take care about the keyboard on your own. Might invite people to try to break out the app :(

PixelAlarm, 2025-03-07:
I see now, works fine.
I misconfigured the max digits and couldn't put in a full mail address.

For input fields with strings, it would be nice to have transparent placeholders that will vanish when entering the field. Otherwise the user always has to delete manually.

mgineer85, 2025-03-07:
Wouldn’t you just leave the default empty and have the label as placeholder?