photobooth-app/photobooth-app #354: Transparent Admin login button

mgineer85, 2024-09-18:
To allow basic functions like
-reboot
-shutdown
implement these functions as button available on non-authenticated page.

To allow access in case admin login button is hidden, have second option to instead make the button transparent and count several clicks before accessing it.

find-usdk, 2024-12-18:
I like the idea about a transparent button, the button could be placed in one of the corners, so that it is easy to hit

mgineer85, 2024-12-19:
This one is implemented in v5.

Will look like following in normal mode:
[image]
Made the button transparent even in default mode because it's less intrusive then. 

If the button is configured to invisible, it's still rendered but with opacity 100%. Also there is a click-counter and the admin page is called only after 5 clicks each within 500ms:
https://github.com/user-attachments/assets/3f437534-56cd-4021-b9d6-310e33b7ece7

find-usdk, 2024-12-19:
Looks great.... Looking forward to all the good improvement in v5.

Another improvement could be the option of having a number-keypad where you could enter a code to access the Admin section. 
This would serve as an alternative to the login field, solving the issue of the touch keyboard not appearing when using kiosk mode.

Example: 
[image]

mgineer85, 2024-12-19:
Yeah, I am hesitant to implement, because it’s just half way. You could enter admin, but then there is still no keyboard to really use it. 
I hope the raspberry on screen keyboard is getting better so it would be actually useful.

on the other side I would expect the shutdown and reboot button to be most useful to access.
These could be located right on the login screen so even normal users could shutdown the system correctly.

commit 1ee2017d (2024-12-19): add option for transparent admin button, fixes #354
closed 2025-01-18 by mgineer85