photobooth-app/photobooth-app #359: Api call show error but worked

Striker62, 2024-09-30:
I'm not sure if it is bug but I try to explain what happend.

I defined a share action to change the frame and take a picture :
the following command is there defined and assigned to the charater "w"

cp /home/pi/photobooth/userdata/frames/wedding.png /home/pi/photobooth/userdata/frames/frame.png && curl -X 'GET' \\ 'http://raspi-photobooth-mini:8000/api/actions/image/0' \\ -H 'accept: application/json

If Ipress "w"
the command work the frame copy worked and the picture was taken.

For me perfect;-)

But I get the following error in the log and on the main screen

ERROR
30.Sep.24 18:15:13	HTTPException: HTTPException(status_code=500, detail="Something went wrong, Exception: Process failed, error Command 'cp /home/pi/photobooth/userdata/frames/wedding.png /home/pi/photobooth/userdata/frames/frame.png && curl -X 'GET' \\ 'http://raspi-photobooth-mini:8000/api/actions/image/0' \\ -H 'accept: application/json' ' returned non-zero exit status 3.")

mgineer85, 2024-09-30:
I am not very sure how chained commands work in linux. cp / curl seem to work, but maybe it's something about the link &&?

Striker62, 2024-09-30:
i tried also only
curl -X 'GET' \   'http://raspi-photobooth-mini:8000/api/actions/image/0' \   -H 'accept: application/json' 
same result ?

Striker62, 2024-09-30:
ok found the issue the correct call is
curl -X GET http://raspi-photobooth-mini:8000/api/actions/image/0 -H 'accept: application/json'
that works without error

closed 2024-09-30 by Striker62

mgineer85, 2024-10-01:
Here is the exit codes overview: https://everything.curl.dev/cmdline/exitcode.html

Striker62, 2024-10-01:
thanks . That is where I found the solution