photobooth-app/photobooth-app #360: Unable to locate package libturbojpeg0

GitNees, 2024-10-08:
**Describe the bug**
Followed the install instructions.

**To Reproduce**
Installed fresh latest Ubuntu on VM:

pasted this command from instructions:
sudo apt -y install ffmpeg libturbojpeg0 python3-pip libgl1 libgphoto2-dev fonts-noto-color-emoji rclone inotify-tools

And I get this in terminal
sudo apt -y install ffmpeg libturbojpeg0 python3-pip libgl1 libgphoto2-dev fonts-noto-color-emoji rclone inotify-tools libjpeg-turbo8-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libturbojpeg0

I googled about this, things are pointing to  libturbojpeg0-dev package, Installed it, but when I start the app I see this:
**Unable to locate turbojpeg library automatically** at then end of the error.

Nees@Ubuntu2:~$ photobooth
Traceback (most recent call last):
  File "/home/Nees/.local/bin/photobooth", line 5, in <module>
    from photobooth.__main__ import main
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/__main__.py", line 12, in <module>
    from .application import app
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/application.py", line 17, in <module>
    from .container import container
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/container.py", line 3, in <module>
    from .services.aquisitionservice import AquisitionService
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/services/aquisitionservice.py", line 16, in <module>
    from .baseservice import BaseService
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/services/baseservice.py", line 6, in <module>
    from .sseservice import SseService
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/services/sseservice.py", line 19, in <module>
    from .mediacollection.mediaitem import MediaItem
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/photobooth/services/mediacollection/mediaitem.py", line 28, in <module>
    turbojpeg = TurboJPEG()
                ^^^^^^^^^^^
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/turbojpeg.py", line 301, in __init__
    self.__find_turbojpeg() if lib_path is None else lib_path)
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Nees/.local/share/pipx/venvs/photobooth-app/lib/python3.12/site-packages/turbojpeg.py", line 968, in __find_turbojpeg
    raise RuntimeError(
RuntimeError: **Unable to locate turbojpeg library automatically**. You may specify the turbojpeg library path manually.
e.g. jpeg = TurboJPEG(lib_path)

mgineer85, 2024-10-08:
Hey, I don’t think you need dev but this: https://launchpad.net/ubuntu/+source/libjpeg-turbo
Does that work?

GitNees, 2024-10-08:
Yes this worked, someone pointed this to me out. Hadn't updated yet. Perhaps script should be changed to install this for Ubuntu. Else other packages are not installed either as the CMD stops.

closed 2024-10-08 by GitNees

mgineer85, 2024-10-08:
Feel free to send pull requests improving the docs. I can test only limited hardware and os so generic Debian/ubuntu is not covered in every detail usually https://photobooth-app.org/setup/installation/#debianubuntu