photobooth-app/photobooth-app #361: DSLR live view still requested with webcam live view specified

Christotheb, 2024-10-12:
**Describe the bug**
Webcam set up as secondary live view backend source. Photobooth still requesting live view from digicamcontrol. Additionally (but not the fault of photobooth-app) digicamcontrol is unable to supply an image via the method that photobooth-app is requesting it while live view is active.
Screenshots below show the digicamcontrol setup. The capture button in the first screenshot does not work when live view is active - only the button labeled "capture" in the live view pane does.

**To Reproduce**
Steps to reproduce the behavior:
1. Have DSLR tethered and webcam set up as secondary backend.
2. Take a photo
3. Photobooth will request live view from digicam
4. Attempt to take another photo after live view is activated
5. Timeout error

**Expected behavior**
Live view should not be requested if a different source for live video is selected. Maybe implement a specific way to choose which device you want live video from and only this device.
Additionally digicamcontrol appears to expect a different "capture image" request for when live view is active on DSLR. 

**Screenshots**
[image]
Capture button that Photobooth is requesting. 

[image]
Capture button that is expected by digicamcontrol while live view is active. 

**System (please complete the following information):**
 - OS: Windows 10 Pro Build 19045
 - Browser Chrome
 - Version 129.0.6668.100 (Official Build) (64-bit) (cohort: M129 Rollout) 
 - DigiCamControl Version 2.1.6.0
 - Photobooth-app v4.0.0

**Additional context**
Log file attached below showing the error 

[photobooth_20241012.log](https://github.com/user-attachments/files/17350478/photobooth_20241012.log)

I'm very new to github so I'm sorry if I have any formatting wrong in this.
I've tried researching if the digicamcontrol issue is a bug but it seems that the software hasn't been updated since last year, and I can't seem to find a support forum that is active.

mgineer85, 2024-10-12:
hi, yeah, the digicamcontrol backend might need improvement. does it work for you when the following line is commented out?
https://github.com/photobooth-app/photobooth-app/blob/5700e3e4cdca81cef6c0071db1739a7f6e98bcdb/photobooth/services/backends/digicamcontrol.py#L197

Christotheb, 2024-10-13:
The behaviour is the same when this line is commented out. 

If I close out the live view manually it works until a capture event is completed, then once it's done it reopens live view again.

mgineer85, 2024-10-13:
You sure you edited the right place? Maybe in virtual env? Because there is only that on function calling the live window

Christotheb, 2024-10-13:
I installed it globally and edited it in the programs folder 

~\AppData\Local\Programs\Python\Python311\Lib\site-packages\photobooth\services\backends which was the only place I could locate digicamcontrol.py on my PC. 

It's entirely possible I've edited the wrong file.

[image]

mgineer85, 2024-10-13:
Sorry to ask, did you restart the app after the change? Could you please send another logging after the change?

Christotheb, 2024-10-13:
No worries! I made the change while the app was offline. Also when I close down the app using CTRL-C the live view switches off. Hopefully this is reflected in the logs too.

[photobooth_20241013.log](https://github.com/user-attachments/files/17355996/photobooth_20241013.log)

mgineer85, 2024-10-13:
Hey, could you try applying this change and see if it fixes it for you?
https://github.com/photobooth-app/photobooth-app/commit/a3e5cb50c1e911e51a83e5f4210606151eaa734b

mgineer85, 2024-10-13:
The other error about `OSError: tjDecompressHeader3(): Could not determine subsampling level of JPEG image` is probably because there is no proper image delivered from digicamcontrol because the liveview interferes, right?

Christotheb, 2024-10-13:
The other error is because digicamcontrol defaults to RAW which is my fault. Nothing wrong with photobooth there, it's just expecting a JPEG.

Christotheb, 2024-10-13:
[photobooth_20241013.log](https://github.com/user-attachments/files/17356058/photobooth_20241013.log)

Still the same issue happening. the moment I run the photobooth app live view activates in digicamcontrol.

I let the program make all its attempts to get a photo while live view is active this time. And then did another attempt where I closed live view and got a successful capture.

mgineer85, 2024-10-13:
Sorry, I forgot the `return`. File should look like this:
https://github.com/photobooth-app/photobooth-app/blob/2b291e35b2dcd7e913dda2a9ec5e32f0a7f4c0ca/photobooth/services/backends/digicamcontrol.py#L177-L179

Christotheb, 2024-10-13:
[photobooth_20241013.log](https://github.com/user-attachments/files/17356080/photobooth_20241013.log)

The webcam live feed doesn't activate and it throws an error when I try to take a capture.

mgineer85, 2024-10-13:
sorry, there was an update to the update already, according to the logs you miss the `return` https://github.com/photobooth-app/photobooth-app/blob/2b291e35b2dcd7e913dda2a9ec5e32f0a7f4c0ca/photobooth/services/backends/digicamcontrol.py#L179

mgineer85, 2024-10-14:
Hey, works with the return included? :)

Christotheb, 2024-10-14:
Hi, I had added the return on line 179 before I tested it.

mgineer85, 2024-10-15:
[image]
But line 180 is executed which means that the return was not hit.

Christotheb, 2024-10-15:
[photobooth_20241015.log](https://github.com/user-attachments/files/17385491/photobooth_20241015.log)

[image]

I forgot to change digicamcontrol from RAW to JPG again but this time it didn't change anything, so I'm wondering if the new error I have is happening before the image is sent. The countdown happens and then I get a little red toast notification saying it didn't work. Sorry if I've done anything wrong

mgineer85, 2024-10-16:
Yeah, python uses indentation for conditionals. Indent by tab or space, mixing is not allowed.
TabError: inconsistent use of tabs and spaces in indentation
tells you about this and the error is at the return line. You need to check the whitespaces to be same. Should be spaces…

Christotheb, 2024-10-16:
I'll redownload the whole file from here after I get home from work this evening.

Christotheb, 2024-10-16:
[photobooth_20241016.log](https://github.com/user-attachments/files/17402860/photobooth_20241016.log)

Yep, all working as intended now!

mgineer85, 2024-10-17:
Great! So the change is included in the next release, closing this. If you have other issues feel free to open another ticket!

closed 2024-10-17 by mgineer85

mgineer85, 2024-11-09:
The next release was published recently: https://github.com/photobooth-app/photobooth-app/releases/tag/v4.1.0

commit a3e5cb50 (2024-10-13): maybe fixes #361