photobooth-app/photobooth-app #363: Collage dimensions don't reflect the settings but a scaled version to fit "full still width"

PixelAlarm, 2024-10-16:
**Describe the bug**
Because of automatic resizing/fitting of images, images in collage and therefore the total collage dimensions scale to the "mediaprocessing: full still width".

My collages (in portrait) get really big (3000x9000) and I cannot append/merge them for printing. (Would result in 12000x9000 in my situation).

**To Reproduce**
1. My "full still width" is set to 3000 - I get a 3000x2000 image that equals 6 Megapixel
2. My collage contains 4 images on top of each other
 - images are set to 1500x1000
 - collage is set to 1500x4500

End result is a 3000 x 9000 collage, since the width is scaled to the "full still width".

**Expected behavior**
The final collage should imo reflect the resolution set in the settings.

**Additional context**
Side note: Sometimes the single images in folder /media/processed/full in total are way smaller than the final collage containing only these images. I think this is based on the recalculation, so just an observation.

mgineer85, 2024-10-16:
Is it possible to share the userdata images and config so I can reproduce more easily?

PixelAlarm, 2024-10-16:
Sure, here you go. Removed the unnecessary parts.
You need something else?

[collage_scaling.zip](https://github.com/user-attachments/files/17396888/collage_scaling.zip)

mgineer85, 2024-10-16:
Hmm, I see. 
To keep collage/image processing uniform would it be an option to 

- change it to scale to height instead of width (but then it breaks when somebody wants to have the collage arranged horizontally)
- check for the longest side to be lower than a configured value (which is currently the width only)
- or just ensure that an image is only shrinked, never enlarged...

I think the second one would be good idea right now.

mgineer85, 2024-10-17:
Currently FULL_STILL_WIDTH is used only in 2 places - this should be easy to replace by a configuration similar to FULL_STILL_LONGEST_SIDE and add a automatic detection which side is the longest.
[image]

I think I'm going to implement above proposal.

There is usually
- ORIGINAL and from these derived the 
  - FULL -> considered as "userspace original" and what the user gets when downloaded as well as used for printing
  - PREVIEW -> used in the gallery when an image is openend
  - THUMBNAIL -> used as thumbnai in the gallery.
FULL/PREVIEW/THUMBNAIL exist as "processed" and "unprocessed" versions for performance reasons - in the UI only processed is used. Processed means pipeline applied, unprocessed not.

PixelAlarm, 2024-10-17:
Thanks for explaining.
I think I still didn‘t get yet, how this is helping with the problem.
For example if I set still_longest to be 3000 (full =3000x2000) as in example above and collage_still_longest to 1500 (full = 1500x1000), the parameter longest_side points to the width and will not help.
Won‘t the result be the same as now?

My idea was to have a fourth version CUSTOM / ACTION with the dimensions from the action-setting that can be directly used for the processing.

PixelAlarm, 2024-10-17:
Just painted a small picture of my situation. Maybe it helps.
[image]

mgineer85, 2024-10-17:
The collage has a canvas width/height setting and also every image can have a defined with/height:
[image]

With that you should be able to define the size you want already.  And the 2x scale should be fixed once the longest side is checked.

PixelAlarm, 2024-10-17:
Perfect, looking forward to the enhancement.

mgineer85, 2024-10-18:
I think this should fix it. I will release it as beta later.
Remind that the old settings are deleted and 3 new added, so if you changed the values, you might need to adjust them again.

commit 55e7abfc (2024-10-18): fix #363
instead resizing full/preview/thumbnail to width it is dynamically checked which side is longer. The longest side will be reduced to the new config values full_still_length/preview_still_length/thumbnail_still_length. The short side is calculated to keep aspect ratio.