photobooth-app/photobooth-app #367: No livestreams outside the action shows

cedric-Swiss, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
Is it possible to activate an image for the basic screen of the application, instead of the liveview, and only activate the liveview, when an action (Photo or collage) is triggered?

**Describe the solution you'd like**
I'd like to be able to display an image on the Photobooth screen, and when an action is triggered, only display the livestream.

PixelAlarm, 2024-11-01:
I use the box the same way. Have a welcome screen and only show the live view when a button is pressed.

My workaround:
1. Place a filename.png inside the web_spa folder.
2. Use this .png as welcome message. For this put this inside the welcome text field in UI settings [image] (keep the styling information as fixed-center etc…

mgineer85, 2024-12-13:
Hey, upcoming version 5 has this feature. 

[image]

closed 2024-12-13 by mgineer85

PixelAlarm, 2024-12-17:
Looking forward to this feature, although for me my workaround is doing well.

Just wondering: What will be displayed, when livestream is disabled?
Will there be some option to define a local image file to be displayed?
Could also be nice to use png as frame/overlay etc.

mgineer85, 2024-12-17:
Right now it would just show the frontend ui text defined. This could be used to display a frame also but is not very intuitive and needs improvement in the future I think.