photobooth-app/photobooth-app #376: Simultaneously DSI Camera and Display

lukashelmke, 2024-12-14:
I have tried using both a DSI camera and a DSI display simultaneously on my Raspberry Pi, but there seems to be a conflict. When both devices are connected, one of them stops functioning properly.
System Details:

Raspberry Pi Model: pi 5
DSI Camera: Raspi-Cam V3 
DSI Display: official 7 inch Raspi touch display

I would greatly appreciate any help or suggestions to resolve this issue. Thank you in advance!

mgineer85, 2024-12-14:
Hi,
this is independent of the PhotoBooth software.
Depending on the screen and which port is connected to which device, you might need to manually define  the screen and camera in your system config.

this could be helpful:
https://github.com/raspberrypi/firmware/blob/24247756bb3e962f0429b048cec5644e4a4253ab/boot/overlays/README#L2882

lukashelmke, 2024-12-15:
> Hi, this is independent of the PhotoBooth software. Depending on the screen and which port is connected to which device, you might need to manually define the screen and camera in your system [...]

thanks!
I changed the config, and it just worked !

closed 2024-12-15 by lukashelmke