photobooth-app/photobooth-app #377: Result page: Reset/extend autoclose-timer on screen press

PixelAlarm, 2024-12-17:
**Is your feature request related to a problem? Please describe.**
No problem, just an idea.
Instead of disabling the timer/autoclose, the timer could just be reset/extended.

When result page is displayed, a timer is going down to autoclose the result page and go back to front page.
When clicking on the result page the timer stops and will never terminate/autoclose. So the X has to be clicked to start another action.

**Describe the solution you'd like**
It would be nice if the autoclose-timer could be reset on screen press.
Or just add some configurable amount of seconds to be added to the timer.

**Purpose**
This way, the booth will ultimately go back to front page when nobody needs the result page anymore and accidentally touched the screen.

mgineer85, 2024-12-17:
I think it would be a good idea to return any time later after interaction automatically again.

But I would not like to have the countdown progressbar shown for the second countdown any more because (at least to me) it would look too busy...

mgineer85, 2025-01-26:
So this is implemented now as follow:

- When on Gallery/Mediaviewer page, the app returns to the index page after 5 minutes (default, can be configured in ui settings).
- When on Index page, the app checks for inactivity and pushes to the slideshow after another configured time.

closed 2025-01-26 by mgineer85