photobooth-app/photobooth-app #379: GPIO Output on countdown for LED Light

find-usdk, 2024-12-18:
It would be great to have a GPIO output that activates when the countdown starts and deactivates again when the picture is taken. 
That way, the output could be used for a ring light or a relay to turn on another lamp.

In this demo video, the button activates both a timer-relay that starts the LED light and activates GPIO 27. 
(The timer controlling the LED light is set to the same duration as the countdown).
https://github.com/user-attachments/assets/96a466e5-2ac9-40a7-8e9f-07d80744ef0b

This ensures that there is enough light so the picture doesn’t turn out dark, and the LED light turns off after the picture has been taken. 

It works fine, but it would be nice to be able to do without the physical button and only use the touchscreen.

mgineer85, 2025-01-27:
Hey, there is an implementation now in main that allows switching a light on/off during countdown.

Would you be able to test it before it is released?

find-usdk, 2025-01-27:
Hi @mgineer85 
Yes, how do i test it? do you have a special update i should run to get it?

mgineer85, 2025-01-27:
I think with the we code issues, I will release a new version very soon that can be tested regularly.

find-usdk, 2025-01-27:
Sounds good, let me know when you release it, then will I test it. 😁

mgineer85, 2025-01-27:
will be added to 5.1.1 released in a few minutes.   in gpio settings you find the light option now

find-usdk, 2025-01-28:
Hi 
I have just tested it, and it works great... 🥇

commit 72baae2b (2025-01-27): add gpio out service to implement and fix #379
closed 2025-01-28 by mgineer85