photobooth-app/photobooth-app #390: Wifi configuration easy ?

Miauto, 2025-01-14:
hi,
I made a box to set up your project, and I have to say it's top.
The photobooth worked smoothly, he was stormed by all the kids, the guests and all the kids 😈, and I have to say that he didn't crash.
Since then he's at every big party, and that's a good memory.
 
They're all right, but when I put the photobooth in a room that has wifi.
It would be nice to be able to configure the wifi easily from the admin interface, without going back to linux with its keyboard.
That would make my friends free-up when I'm not here.

2 ideas in mind:
      1. set up a menu as if to configure your phone for wifi.
      2. Use the camera to read a QRCODE

What do you think of this idea?
And do you think about that?

Nice job.
Have a good day.

mgineer85, 2025-01-14:
I like both ideas, but it might not be so easy to integrate because it needs to work on windows and linux.
There is pywifi which could be used but it is not maintained since 7 years.

I think the overall setup of a box on an event can be streamlined a bit more, this is one part of it.

I leave this open, would be happy if somebody would come up with ideas on how to implement it actually :)

mgineer85, 2025-01-14:
Also, happy the box works well for you ✨

Would you be interested to show some more details of your project and describe the box?
You could post here https://github.com/photobooth-app/photobooth-app/discussions/categories/show-and-tell if you like to :)

mgineer85, 2025-02-15:
Hi, thank you for the suggestion.

I would love to ease the wifi config more, but there are currently no platform agnostic tools available to do that. 

So I think it will not happen in the near future, because the app shall work on windows and linux in the same way.

closed 2025-06-03 by mgineer85