photobooth-app/photobooth-app #400: Implement "AI Art Filters" by using local or external services

soerennb, 2025-01-22:
AI models open a new world for image manipulation and style transfer. Instead of just changing the colors/brightness/contrast they allow to completely change the style and even the faces of people. 
The photobooth admin should be able to switch from the Pilgrim2 filters to an AI powered service. As there are tens & hundreds of service providers out there, this part should support **plugins**/provider modules. The Photobooth-App doesn't have a plugin system at the moment. This allows to have a unified interface, but different providers, _each with its own configuration (like URL, auth etc.)_. 
It should also be possible to use local services (e.g. [stable-diffusion-webui-forge](https://github.com/lllyasviel/stable-diffusion-webui-forge)).
The plan is to process a single image by sending it via API to a img2img service and showing the result to the user. This part should be handled by the server. The available filters should be in the same place as now (gallery item view). These "Filters" are nothing else than prepared prompts and settings for Stable Diffusion or other services accompanied by a picture that shows the style.
I will start working on a first implementation of filters using the API of stable-diffusion-webui-forge now.

mgineer85, 2025-01-22:
You are familiar with a service and tried some prompts? I did not try very much recently and had only little luck getting nice results.
You have some examples? Would love to try it separately once 🙂

soerennb, 2025-01-29:
I have worked out a solution for using a locally-running Stable Diffusion Model. Do you have the chance to install [Stable Diffusion WebUI-Forge](https://github.com/lllyasviel/stable-diffusion-webui-forge?tab=readme-ov-file#installing-forge) (on a Windows/Linux machine with a dedicated GPU)? Then I can give you some examples to try out.
But I will also provide a pull request, so you can review the changes. Some new configuration parameters are needed ;-).

mgineer85, 2025-01-29:
I have no gpu unfortunately.

What is the processing time on your computer?
I think in the end it might only be usable online on a pi?

soerennb, 2025-01-29:
> I have no gpu unfortunately.
> What is the processing time on your computer?

Processing time on a RTX 4070 is ~10 seconds

> I think in the end it might only be usable online on a pi?

Yes, you need to use an external (or online) service on a pi.

PixelAlarm, 2025-01-30:
Really curious about this idea. Could be a killer feature when coupled with a fast online-service.
My connection should be sufficient for this :)

Keep it up!

mgineer85, 2025-01-30:
> > I have no gpu unfortunately.
> > What is the processing time on your computer?
> 
> Processing time on a RTX 4070 is ~10 seconds
> 
> > I think in the end it might only be usable online on a pi?
> 
> Yes, you need to use an external (or online) service on a pi.

Would it work with the pi 5 npu extension also?
https://www.berrybase.de/raspberry-pi-ai-hat-26-tops-hailo-8-accelerator?utm_source=google&utm_medium=cpc&gad_source=1&gclid=Cj0KCQiA4-y8BhC3ARIsAHmjC_Hg4fvzR2KAk3ZE3VO1tmgpHQeS6xqov57U1_MhSgwzckEKcKdpnfYaAgSJEALw_wcB
I have this available…

soerennb, 2025-01-30:
This should be possible, but I'm not sure if this is practical for using it from photobooth-app ;-)
https://www.youtube.com/watch?v=HgIMJbN0DS0&t=364s

mgineer85, 2025-02-14:
Hey @soerennb , out of curiosity, did you make some examples work?  I would be interested in adding this somehow to the app. Maybe the new plugin system could be helpful here...

mgineer85, 2025-03-03:
Hey @soerennb ,

I just created a branch based on latest code and took over the files needed manually. I think more should not be needed. 
It's just copied and will not work yet.

I still need to understand a bit more how the filter works, what configuration is needed and so on.

If you want to contribute, have a closer look to dev_filterstable:
https://github.com/photobooth-app/photobooth-app/compare/main...dev_filterstable

Otherwise I will continue in the next days/weeks when I have time to integrate this :)

I will close the old pull request, if there is any discussion needed, we can use this issue.

mgineer85, 2025-03-03:
Hello @soerennb , 

I got the service running using getimg.ai but I get nowhere near as nice images as you get.
You have an idea what I am missing? Probably some settings could be improved, or is this expected with the getimg.ai models?  Could we have the same as you use locally?

https://github.com/photobooth-app/photobooth-app/compare/main...dev_filterstable

soerennb, 2025-03-03:
I even have trouble getting the same results locally via the Python Web API at the moment ;-)

I hope that the Runway API provides better quality images. The runway python SDK is buggy, so I need to manually handle the API requests. But the service looks promising over all.

mgineer85, 2025-03-03:
Ok, will fiddle around more.
Just thought maybe the implementation for the local service was better than the first draft using online services? 
They would run the same models, don’t they?

mgineer85, 2025-03-05:
I just tested the clay filter until now and do not get stable/consistent results.

I looked around and like these filter very much: 
https://openart.ai/apps/ai-filter

I would like to understand how that works and run these filter on the runware api. Could that work?

soerennb, 2025-03-05:
Openart.ai doesnt't provide an API, it seems. "Styles" are nothing else than different prompts to a model like Stable Diffusion. The model alone doesn't provide enough quality, so that's why I use 3 Controlnets to guide the image generation process. That's not possible with getimg.ai.

mgineer85, 2025-03-05:
it seems to be possible with runware.ai , but I am not able to configure it to get good consistent results. They have a playground for free .

If anybody has working settings for some filter types, I can add these easily

soerennb, 2025-03-08:
I've also played around with runware. In my experience the Flux models are not  as good as stable diffusion models for img2img. I couldn't yet find settings/workflows to achieve aesthetically pleasing results, but keep trying with my remaining test credits ;-).

mgineer85, 2025-09-13:
Found this one, could be helpful to find settings and tinker around: https://github.com/ansemjo/diffusionbooth

PixelAlarm, 2025-11-10:
Hi, I did some img2img manually recently and have been curious, if there is any activity on this topic :)

I saw some code merges some time ago, but right now the plugin isn‘t available, right?

Would be awesome to have something into this direction to play around with.

mgineer85, 2025-11-10:
I did not find stable working pipeline yet, that could work local and online.

PixelAlarm, 2025-11-16:
> Haha, nice it added a kitchen to my virtual video 🤣
> 

I tinkered with the given prompt from here and the prompt from soerennb had the kitchen background in it 😅

@mgineer85 do you have the plugin with working runware.ai at hand?
I somehow lost my working version and don’t want to start over again 😭