photobooth-app/photobooth-app #401: Share via QR-Code Version v5.0.0rc3

Striker62, 2025-01-24:
I try to download a photo to my iphone via QR_Code share.

The url wher I can reach my Booth form my pc is : http://raspi-photobooth-mini:8000/
This Url I can reach from my Iphone as well.
The path for the Qr-Code share in config is : http://raspi-photobooth-mini:8000/media/processed/full/{filename}
When I scan the QR-Code and start the browser on the iphone 
the Result is "Zeitüberschreitung bei der Anforderung" in my browser on the I phone.

I forward the Url from the Iphone browser to my pc :http://raspi-photobooth-mini:8000/media/processed/full/80d591fb-306f-42f3-9ebe-107dd1f7c6b0

When I try this on my pc I get {"detail":"Not Found"}

Can it be the case that this is due to the new sqldb ?

I also try to print on my HP printer 
Config   lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo {filename}
I get an error see log 

Share photo via telegram is working fine.
Config  /usr/local/bin/send_photo_to_tg  {filename} "test"

I did the upgrade from version 4.0.0  

2025-01-24 14:12:06,154 [   ERROR] Process failed, error Command 'lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo /home/pi/photobooth/media/processed_full/20250124-123938-447490.jpg' timed out after 6 seconds (share.py:24)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/photobooth/services/share.py", line 106, in share
    completed_process = subprocess.run(
                        ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 550, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 2060, in _communicate
    self._check_timeout(endtime, orig_timeout, stdout, stderr)
  File "/usr/lib/python3.11/subprocess.py", line 1251, in _check_timeout
    raise TimeoutExpired(
subprocess.TimeoutExpired: Command 'lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo /home/pi/photobooth/media/processed_full/20250124-123938-447490.jpg' timed out after 6 seconds

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/photobooth/routers/api/share.py", line 18, in _share
    container.share_service.share(mediaitem, index)
  File "/home/pi/.local/lib/python3.11/site-packages/photobooth/services/share.py", line 122, in share
    raise RuntimeError(f"Process failed, error {exc}") from exc
RuntimeError: Process failed, error Command 'lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo /home/pi/photobooth/media/processed_full/20250124-123938-447490.jpg' timed out after 6 seconds
2025-01-24 14:12:06,158 [   ERROR] HTTPException: request.url=URL('http://raspi-photobooth-mini:8000/api/share/actions/80d591fb-306f-42f3-9ebe-107dd1f7c6b0/1') exc=HTTPException(status_code=500, detail="Something went wrong, Exception: Process failed, error Command 'lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo /home/pi/photobooth/media/processed_full/20250124-123938-447490.jpg' timed out after 6 seconds") (application.py:87)
2025-01-24 14:12:06,159 [    INFO] 192.168.2.140:65205 - "GET /api/share/actions/80d591fb-306f-42f3-9ebe-107dd1f7c6b0/1 HTTP/1.1" 500 (h11_impl.py:473)
2025-01-24 14:12:10,314 [    INFO] 127.0.0.1:37056 - "GET /media/preview/8eb3c7c9-752a-4a20-b513-4d1abc9560bd HTTP/1.1" 206 (h11_impl.py:473)
2025-01-24 14:12:16,708 [    INFO] 192.168.2.140:65239 - "GET /media/thumbnail/8eb3c7c9-752a-4a20-b513-4d1abc9560bd HTTP/1.1" 206 (h11_impl.py:473)
2025-01-24 14:12:23,928 [    INFO] 192.168.2.140:65252 - "GET /api/admin/config/schema?schema_type=dereferenced HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:12:23,979 [    INFO] 192.168.2.140:65252 - "GET /api/admin/config/currentActive HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:12:34,464 [    INFO] 192.168.2.140:65282 - "GET /media/preview/80d591fb-306f-42f3-9ebe-107dd1f7c6b0 HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:12:38,149 [    INFO] 192.168.2.140:65282 - "GET /assets/AdminLogsPage-D1-uSfrd.js HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:12:38,151 [    INFO] 192.168.2.140:65291 - "GET /assets/QMarkupTable-Dm3Hn0z0.js HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:12:50,308 [    INFO] 127.0.0.1:39408 - "GET /media/preview/ea45a982-f628-4541-9d64-3a81edf462cb HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:13:05,284 [    INFO] 127.0.0.1:36574 - "GET /media/preview/ea45a982-f628-4541-9d64-3a81edf462cb HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:13:15,278 [    INFO] 127.0.0.1:37568 - "GET /media/preview/80d591fb-306f-42f3-9ebe-107dd1f7c6b0 HTTP/1.1" 200 (h11_impl.py:473)
2025-01-24 14:13:39,800 [    INFO] getting latest logfile: log/photobooth_20250124.log (debug.py:29)

PixelAlarm, 2025-01-24:
For the QR-Code issue:
Since 5.0.0_rc1 there is a new folder structure on my end.

It's now .../media/processed_full instead of .../media/processed/full

Striker62, 2025-01-24:
I've changed the config 
http://raspi-photobooth-mini:8000/media/processed_full/{filename}

result
http://raspi-photobooth-mini:8000/media/processed_full/8465ce7d-0019-4c89-921a-924cb011a611

That give me :

{"detail":[{"type":"enum","loc":["path","dimension"],"msg":"Input should be 'full', 'preview' or 'thumbnail'","input":"processed_full","ctx":{"expected":"'full', 'preview' or 'thumbnail'"}}]}

PixelAlarm, 2025-01-24:
Oh I see.
I don‘t know how the files are served then.

Just saw, that the structure changed and thought it could help.

Striker62, 2025-01-24:
ok now it works ;-)

Printing also working for what ever reason .
I only did the pip with the new official release wich is the same as i did some hours ago  v5.0.0rc3

So I'll kepp on testing .
But so far again great job.

mgineer85, 2025-01-24:
For the other error about the share to printer, there is an timeout. That means the command did take longer than 6 seconds before it finished and is considered as error.

Maybe the printer was not yet ready ?

raise TimeoutExpired(
subprocess.TimeoutExpired: Command 'lp -d HP_ENVY_5640_series_1AB223 -o media=Photo4x6 -o landscape -o InputSlot=Photo /home/pi/photobooth/media/processed_full/20250124-123938-447490.jpg' timed out after 6 seconds

Striker62, 2025-01-24:
Yes could be. Thanks.
I tested all features I used and setup for my sons wedding .
Every thig is working fine and better ;-)