photobooth-app/photobooth-app #405: Frame per action

mgineer85, 2025-01-24:
> Made directly the Update, there is one thing, i found out: for the framepath, you have to add userdata/ in the beginning of the path, otherwise, it does not work. (could cause problems, because an error appears.
> 
> For this functionality i was hoping, that the frame changes in liveview to the frame, which is used for the action.
> for example an action with this frame:
> 
> [image]
> 
> The users need to know, where to put the head, that the frame is exactly over theirs. so during the countdown, it has to be this frame.
> 
> but for other actions, it shoould not be there, so in standart live stream again the normal Frame, which could be this:
> 
> [image] 

 _Originally posted by @McKaiver in [#203](https://github.com/photobooth-app/photobooth-app/issues/203#issuecomment-2612857731)_

mgineer85, 2025-01-26:
This is implemented now:

https://github.com/user-attachments/assets/bb223740-fb6f-4a73-a41d-176ca3a435c2

Since the processsing in the backend and the display in the frontend work slightly different, there can be disalignment between the processed image and the live preview.
When the resolutions of the frames and the camera align well, there is no issue.

closed 2025-01-26 by mgineer85

McKaiver, 2025-01-26:
well, that is perfect!!!
from which version on and when will it be live XD

mgineer85, 2025-01-26:
I would call it 5.1 and somewhat today

mgineer85, 2025-01-26:
https://github.com/photobooth-app/photobooth-app/releases/tag/v5.1.0

McKaiver, 2025-01-26:
Ok, made the first tests, It works exactly, as i imagined. PERFECT!

mgineer85, 2025-01-26:
Awesome 👍🏻 

I think at one point there might be adaptions needed to align backend/frontend processing but for now I think it's good 💃🏻