photobooth-app/photobooth-app #406: Custom URL on Qr code not working like V4 on V5

zendre4, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
On V4 {filename} is the name on the filesystem, on V5 {filename} is the identifier. I use the sftp sync script to sync my files in a website, and my files have the same name like filesystem. So in v5 my qr Share not working anymore. 

I make a pull request for this

mgineer85, 2025-01-25:
I see how this makes sense.
I’ll review it asap and merge it.
Thank you!

zendre4, 2025-01-25:
Thanks !  If I can help (specially on frontend) tell me 😄

zendre4, 2025-01-25:
> You use the unprocessed version also to share? I kind of would remove it
> because I think nobody would ever would provide the unprocessed version
> since it looks not as displayed to the user...

No I don't ... I have implemented this because choice because I don't know
if someone use this or not.

zendre4, 2025-01-25:
Would'you like I remove it?

mgineer85, 2025-01-26:
Initially I thought it might be nice but it's just overhead.
For the PHP qr share solution I removed it now and for the manual version you still could point to the original folder.

mgineer85, 2025-01-26:
Updated code will be shipped soon, closing this for now.

mgineer85, 2025-01-27:
fixed in 5.1.1 released in a few minutes.

closed 2025-01-27 by mgineer85

zendre4, 2025-01-27:
In fact, in my correction I was looking for the last part of the url with "/" as on Linux and in version 5.1.0 it's "\" as like Windows. I was going to suggest a fix but I see that @mgineer85  is more responsive 🤣. So Thanks @mgineer85

mgineer85, 2025-01-27:
For me this fixed it on both platforms: https://github.com/photobooth-app/photobooth-frontend/commit/4e0892b5d0e8dbd6f6035fbcb6e1e9ad98c658cb