photobooth-app/photobooth-app #408: Adjustable glass-effect-color / Primary color has no effect on button-coloring anymore

PixelAlarm, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
I really like the glass effect. Well done!
(I used to implement it myself via the private.css)

Since this effect is introduced, the primary UI color has no effect on the button styling anymore.
The buttons always seem to be blueish.

It would be really nice to be able to adjust the color in some way.

**Describe the solution you'd like**
Option 1: Introduce third color picker for the button color.
Option 2: Use primary color as basis for the new glassy buttons.

Would also be nice to be able to adjust the strength of the glass effect / overall transparency of the button

**Describe alternatives you've considered**
achievable via private.css

**Additional context**
Again, thanks for v5 and all your work.

mgineer85, 2025-01-26:
Will be fixed by above commit. It's will now also respect the dynamically set variables and so change the color :)

closed 2025-01-26 by mgineer85

mgineer85, 2025-01-26:
Updated: https://github.com/photobooth-app/photobooth-app/releases/tag/v5.1.0