photobooth-app/photobooth-app #409: add the action buttons next to the result screen

mgineer85, 2025-01-26:
Since the new Layout is now released, I thought a bit about this idea.

Option 1: 
For my setup it would be possible to add the action buttons next to the result screen options.
I mostly only use 2-3 actions and only allow delete (and sometimes printing) on the result screen.
So it would be only up to 5 buttons in a row on the bottom (picture, collage, print, delete, filter)

Option 2:
Another option would be to move the result screen options (share, delete, filter) to the right under the qr button.
The action buttons (picture, collage) could be displayed at the same spot as on the frontpage.
That way the two "sections" would be separated and hopefully won't confuse the users.

_Originally posted by @PixelAlarm in https://github.com/photobooth-app/photobooth-app/discussions/378#discussioncomment-11954728_

zendre4, 2025-01-31:
Good idea. Personally I prefer options 2. I can help if needed

mgineer85, 2025-01-31:
Option 2 is good. Only the filter sidebar will hide the share tab then. Could be ok, but maybe it could be solved better.

zendre4, 2025-01-31:
Ah yes just, I didn't think of the filters. Personally, I don't let people to change the filters. 

And what would you think of putting the print and delete buttons on the left side? Like centering them so they're not directly under the back button. Otherwise, if we stay on the right, we could imagine move a litle bit the buttons if the filter bar is open.

PixelAlarm, 2025-02-10:
I also keep the filters deactivated right now, but I see that they have to be considered.

So possible options that could be considered:

Option 1: Since the filterbar covers the right sight, the result screen actions (delete, filters, share) could be displayed on the left under the back button and the picture actions on bottom middle like on the frontpage.

Option 2: Display result actions on the right and move the filterbar to the bottom or unter the Back Button.
The back button should be visible all the time in my opinion. The image actions could rather be covered when filterbar is active.

Wihoui, 2025-04-29:
Hello,
Regarding this page, is it possible to reduce the viewing time before returning to the home screen? 
Is this configurable?

Thank you, your App is so awesome !

PixelAlarm, 2025-04-29:
> Hello, Regarding this page, is it possible to reduce the viewing time before returning to the home screen? Is this configurable?
> 
> Thank you, your App is so awesome !

There are two options:

1. Uisettings „Show frontpage timeout“
Will return to homescreen after time in minutes when the result page was clicked and timer was paused

2. Uisettings „autoclose new items arrived“
Will return to homescreen after time in seconds when nothing is clicked on result page. Upper bar will indicate remaining time.

mgineer85, 2025-04-29:
Yes in uisettings-> AUTOCLOSE_NEW_ITEM_ARRIVED

Wihoui, 2025-04-29:
> Yes in uisettings-> AUTOCLOSE_NEW_ITEM_ARRIVED

Yes Perfect !