photobooth-app/photobooth-app #414: Colors for trigger actions buttons

zendre4, 2025-01-31:
**Is your feature request related to a problem? Please describe.**

I'd like to be able to assign a custom color to the action buttons...  I find this is more visual for visual person like me. In addition, it lets you match a color to an external physical button.

**Describe the solution you'd like**

[image]

zendre4, 2025-01-31:
### PR: 

**Backend:**

https://github.com/photobooth-app/photobooth-app/pull/415

**Frontend:**

https://github.com/photobooth-app/photobooth-frontend/pull/100

mgineer85, 2025-01-31:
We had this in the custom theme section, but this is of course more comfy.

Ref: https://photobooth-app.org/reference/customizetheme/#example-colorize-the-buttons-on-the-frontpage

Will merge it this weekend.

Thank you!

zendre4, 2025-01-31:
Oh, sorry, I didn't see that option. Thanks if you agree to merge. This isn't pressing for me, but I thought it would be cool to be able to do it.

mgineer85, 2025-02-09:
[image]

merged, thx!

Will be released in v6 together with the plugin extensions

closed 2025-02-09 by mgineer85

zendre4, 2025-02-10:
Thanks @mgineer85 !

I've had a look at the plugins branch, and I'm looking forward to using it. 😈