photobooth-app/photobooth-app #416: Customize the Theme > Adjust the Livestream not working anymore

PixelAlarm, 2025-02-01:
**Describe the bug**
Since introduction of the Blurredbackground option, the fitting of the stream to the window isn't working anymore:

```
#preview-stream {
  background-size: cover; /* default: contain to avoid cropping */
  transform: scale(1.5); /* default: 1.0 to apply no zoom. 1.5 zooms in, smaller values than 1.0 usually make no sense */
}
```

**To Reproduce**
(Optionally) Disable Livestream Blurredbackground and Enable Livestream Frameoverlay.
Set preview stream to "background-size: cover;" via private.css

**Expected behavior**
The video stream used to be fitted to the screen size. Now the setting has no effect.

**System (please complete the following information):**
Tried in chromium, safari, firefox.

**Additional context**
When using "Enable Livestream Frameoverlay" and forcing "background-size: cover;" the live frameoverlay might be in a different position than the processed image.

For me, this is ok, because I mostly use the live frameoverlay to decorate the screen.
(Disabling the mirror effect would be nice in my case.)

If the frame overlay is used for positioning, this might create some trouble.
An option would be to set the frame overlay to fit with the live stream maybe.

mgineer85, 2025-02-03:
I see, the structure of the html/css heavily changed in the latest version.
I check if we can have another mode selectable.

PixelAlarm, 2025-02-03:
The element that has to be modified is now #overlay-image like so:

#overlay-image {
  width: 100vw; /* 100% der Bildschirmbreite */
  height: 100vh; /* 100% der Bildschirmhöhe */
  background-position: center; /* Das Bild wird zentriert */
  background-repeat: no-repeat; /* Keine Wiederholung des Hintergrunds */
  background-size: cover; /* Bild wird so skaliert, dass es den gesamten Bereich abdeckt */
}

Using the element #overlay-wrapper the frame is scaled according to the video and would allow for positioning.

I wasn't able to mirror the frame independent from the live stream though.
Would appreciate, if separation of the mirroring for frame and video would be possible.

mgineer85, 2025-02-14:
The css is update in the docs now.

The stream and frame are currently in the same element so the mirror effect cannot apply separate unfortunately.
This is needed to align frame and stream well to each other with the correct aspect ratio.

closed 2025-02-14 by mgineer85

PixelAlarm, 2025-03-12:
Hi,
recently found out, that the given solution in the docs is only partly working.

It only works, if a frame overlay is activated.
1. With activated frame, the element can be addressed with id: overlay-image
--> Solution in docs will work
 
2. Without a frame, the preview element can be addressed with id: stream-image--> No element overlay-image
--> Another solution has to be used:

#stream-image {
  height: initial; /*reset parameter*/
  max-height: initial; /*reset parameter*/
  width: 100vw;
  background-size: cover;
  transform: scale(1.0);
}

Maybe there is some way to address a higher level id and achieve the same results with one adjustment. For now I added both adjustments to the private.css, since they won't interfere with each other.