photobooth-app/photobooth-app #417: Photobooth app does not autostart in LABWC

WebfinityDesign, 2025-02-03:
After updating the OS I got a message about switching to LABWC. After switching to LABWC the photobooth app does not autostart anymore. Is it better to keep wayfire or switch to LABWC and adjust some settings?

mgineer85, 2025-02-03:
The service starts, but with labwc the autostart works different.

The instructions might not cover labwc yet, so you might need to look up in forums currently how to solve.

mgineer85, 2025-02-04:
Added to the description now:
https://photobooth-app.org/setup/installation/#labwc

closed 2025-02-04 by mgineer85

rriskus, 2026-10-01:
Hey @mgineer85  just wondering if this section is gone

https://photobooth-app.org/setup/installation/#labwc

Seems empty now and autostart docs doesn't fully explain where to create files for kiosk mode.

I might be missing something but just wondered if something has changed. I also never messed around with these settings but it would be good to have a better explanation for first time setup.

mgineer85, 2026-10-01:
Hi, I removed information for legacy pi os systems done time ago because I assume new systems always use the latest images from raspberry pi.

The changes have been made in above linked commit: https://github.com/photobooth-app/photobooth-docs/commit/5b000d35537c2a037cd792b06b9a2fc209ef814e

Maybe you find helpful information there?
I am going to double check if there is some information missing after moving to another docs platform..