photobooth-app/photobooth-app #429: Translate notification messages

pierrelesek11, 2025-02-12:
Some notifications are not yet included in the language file, it would be great if these could be translated like the rest of the application.

Example below:
[image]

mgineer85, 2025-03-31:
Unfortunately, some of these messages are sent from the backend directly , so there is no translation currently as only the ui is translated.
So it will take bit more effort to translate.

mgineer85, 2025-04-05:
So, basically it is about to translate all the notifications wrapped in the `SseEventFrontendNotification` object. 
Overall it's only 10 messages until today, so it can be done, just need to find a good way.

mgineer85, 2025-04-06:
So, the user-facing messages are now translateable and pretranslated using crowdin.

If errors occur (which shall be rare), these messages will not be translated.

Feature will be included in v7 release later.

closed 2025-04-06 by mgineer85