photobooth-app/photobooth-app #431: dynamic switch hi/lo res for v4l/cv2

mgineer85, 2025-02-13:
Hi, did you tinker around a bit and could solve it?

I tested with a 4k camera laying around here following script:
```python
import time

from linuxpy.video.device import Capability, Device, VideoCapture

MODES = {
    "auto": None,
    "mmap": Capability.STREAMING,
    "read": Capability.READWRITE,
}

def run(frame_size=(640,480),capture_frames_nb:int=100):
    width, height = frame_size

    with Device("/dev/video0") as device:
        print(device.info.card)
        capture = VideoCapture(device, size=2)
        capture.set_fps(10)
        capture.set_format(width, height, "MJPG")
        start = last = time.monotonic()
        last_update = 0
        with capture:
            fmt = capture.get_format()
            fps = capture.get_fps()
            print(f"Starting capture {fmt.width}x{fmt.height} at {fps} fps in {fmt.pixel_format.name}")
            for frame in capture:
                new = time.monotonic()
                fps, last = 1 / (new - last), new
                if new - last_update > 0.5:
                    elapsed = new - start
                    size = len(frame) / 1024
                    print(
                        f"Frame: {frame.frame_nb:>8} Size: {size:>8.1f} Kb | Elapsed: {elapsed:>8.1f} s | Rate {fps:>8.1f} fps",
                        end="\r",
                    )
                    last_update = new

                if frame.frame_nb>=capture_frames_nb:
                    break
try:
    run((640,480))
    run((4192,3104),capture_frames_nb=1)
    run((640,480))
except KeyboardInterrupt:
    print("stopped")
```

It seems the camera can switch quite fast between low resolution and high resolution on my system.

We could implement a dynamic switch, but need to consider also:

- affects closing the camera the focus?
- is the FOV different for different resolutions?
- is the camera tuning flushed when the camera switches between modes and would need some frames to settle? could the old settings be transferred in the high res mode?
- probably more, when looking into the details.

I think it's worth to implement something like that if it works reliable because webcams will become more high resolution anyways and so we can utilize them better in the future :)

_Originally posted by @mgineer85 in https://github.com/photobooth-app/photobooth-app/discussions/423#discussioncomment-12189785_

mgineer85, 2025-02-13:
script used for testing.

the switch from lores to hires is usually 0.4s which is acceptable low for a user not to note.
Would be interesting what numbers you receive @tomtothet ?

```python 

import time

from linuxpy.video.device import Device, VideoCapture

lores = (640,480)
hires = (4192,3104)

def save(filename, data:bytes):
    with open(filename, "wb") as file:
        file.write(data)

def run_lores_stop_capture_run_hires():

    with Device("/dev/video0") as device:
        print(device.info.card)
        print(device.info.formats)
        print(repr(device.info.capabilities))

        capture = VideoCapture(device, size=2)
        capture.set_fps(25)

        # lores start
        capture.set_format(lores[0], lores[1], "MJPG")
        start = last = time.monotonic()
        last_update = 0
        with capture:
            fmt = capture.get_format()
            fps = capture.get_fps()

            print(f"Starting capture {fmt.width}x{fmt.height} at {fps} fps in {fmt.pixel_format.name}")
            for frame in capture:
                new = time.monotonic()
                fps, last = 1 / (new - last), new
                if new - last_update > 0.5:
                    elapsed = new - start
                    size = len(frame) / 1024
                    print(
                        f"Frame: {frame.frame_nb:>8} Size: {size:>8.1f} Kb | Elapsed: {elapsed:>8.1f} s | Rate {fps:>8.1f} fps",
                        end="\r",
                    )
                    last_update = new

                if frame.frame_nb>=100:
                    save('frame_before_mode_change.jpg',bytes(frame))
                    break

        tms=time.monotonic()
        # hires capture 1 frame
        capture.set_format(hires[0], hires[1], "MJPG") # set_format works only if camera is closed!
        start = last = time.monotonic()
        last_update = 0
        with capture:
            fmt = capture.get_format()
            fps = capture.get_fps()
            print(f"Starting capture {fmt.width}x{fmt.height} at {fps} fps in {fmt.pixel_format.name}")
            for frame in capture:
                size = len(frame) / 1024
                print(                    f"Frame: {frame.frame_nb:>8} Size: {size:>8.1f} Kb"                )
                last_update = new

                # just one hi res capture
                tme=time.monotonic()
                save('hires.jpg',bytes(frame))

                break

        print(f"--> took {round(tme-tms,1)}s to switch from lores to hires and get a frame from cam")
        
        # lores continue
        capture.set_format(lores[0], lores[1], "MJPG")
        start = last = time.monotonic()
        last_update = 0
        with capture:
            fmt = capture.get_format()
            fps = capture.get_fps()

            print(f"Starting capture {fmt.width}x{fmt.height} at {fps} fps in {fmt.pixel_format.name}")
            for frame in capture:
                new = time.monotonic()
                fps, last = 1 / (new - last), new
                if new - last_update > 0.5:
                    elapsed = new - start
                    size = len(frame) / 1024
                    print(
                        f"Frame: {frame.frame_nb:>8} Size: {size:>8.1f} Kb | Elapsed: {elapsed:>8.1f} s | Rate {fps:>8.1f} fps",
                        end="\r",
                    )
                    last_update = new

                if frame.frame_nb>=100:
                    break
        
        print()

try:
    run_lores_stop_capture_run_hires()
    # run((4192,3104),capture_frames_nb=1)
    # run((640,480))
except KeyboardInterrupt:
    print("stopped")
```

mgineer85, 2025-02-13:
Für einen ersten test ist es in der Software im main stream implementiert.

Scheint soweit zu funktionieren, im gesamten dauert es 1 sekunde ab anforderung des hochauflösenden fotos bis es dann fertig gespeichert ist.

https://github.com/user-attachments/assets/13a889af-06cf-4a36-8c05-f371b19c5d1a

Hier auf einem Pi5 und angezeigt auf dem normalen PC, aber ich habe wenig bedenken, dass der Pi4 das nicht schafft.

mgineer85, 2025-02-13:
Es gibt nun 2 Auflösungen.

[image]

Ggf. sollte das Umschalten abschaltbar sein.

mgineer85, 2025-02-13:
There is a test version available for anyone who is brave :) 
https://pypi.org/project/photobooth-app/6.0.0a1/

tomtothet, 2025-02-14:
Hi Michael. Thank you so much for getting this done so quickly! I installed v6.0.0a2 today using "pipx install --suffix=@6.0.0a2 'photobooth-app==6.0.0a2'" so that I would be able to go back to the stable release, just in case. However, I didn't face any noticeable issues. I'm using V4l2 backend with my Link 2C and set preview res to 720p and still res to 4k. The overall performance is so much better! It feels like you've made additional changes affecting this positively, did you? Whatsoever, the perceived performance is much better compared to v5.1.1 even though the live view res is the same at 720p. This is also underpinned by the visibly lower CPU utilization shown in the admin panel dashboard. It's somewhere around 35% now, compared to 50% and over before, even in idle state. It seems that you didn't have to reduce the fps rate, even though this might be worth a separate setting in order to limit this for preview for cams with 60 fps natively, respectively for slower hardware.

Even though I don't notice any issues, one warning and 2 errors have been logged starting up v6.0.0a2: "device raised exception (maybe lost connection to device?). I don't have too much time today to keep playing around today. I'll see if this keeps popping up continuing my tests tomorrow. For now, please find the log file attached. 

[log_20250214_1751.txt](https://github.com/user-attachments/files/18802597/log_20250214_1751.txt)

I also performed a first test with the Pi Cam 3 today. I can already tell that there's a tremendous difference in the image quality, especially with not ideal light conditions. I'll try to capture comparison pics tomorrow. Looking at the great feature you've just implemented, I will for sure go with version 6 and the Link 2C!

mgineer85, 2025-02-14:
On first sight it seems the fps cannot be set. Will check how to mute that …

mgineer85, 2025-02-15:
I will release another version soon that catches the error if the FPS rate cannot be set. Could you please send a full log with the new version then to confirm it looks well?

Also I added another configuration item ``flush_number_frames_after_switch``:

> After switching the format, to high resolution, the camera might need some frames to accomodate to the light again. Use the lowest numer of frames that gives the same image as before in preview mode. If too low, images might apper darker or lighter than expected.

I tested with 5 frames and it gives me better result for only little more time to shoot the image.

mgineer85, 2025-02-15:
https://pypi.org/project/photobooth-app/6.0.0b1/

tomtothet, 2025-02-15:
I made 2 test fotos this morning. The horizontal angle is a little different as the Link 2C is on top of my screen and the Pi Cam 3 is standing directly on the surface of my desk. I guess the difference is so clear that no one's going to argue that the angle is causing the difference. The cams are facing side light from the window which is roughly 75° to view axis. The Pi Cam 3 is clearly struggling with the conditions, whereas the Link 2C does an awesome job here in my opinion. The colors are vivid to over-saturated. This is the basic setting of the cam. For a photobooth I would argue that this is rather what people expect than a ultra-realistic foto. Find the pics attached.

I'll install 6.0.0b1 next and will let you know.

[image]
[image]

tomtothet, 2025-02-15:
One more thing I forgot to mention. The Link 2C seems to comensate the "mirror effect" by default. The "Livestream mirror effect" toggle is off in the configuration. While this is perfect for live view as it's more natural, the actual foto taken should ideally not be mirrored. I experimented with the "Orientation" setting of the V2l2 backend, but it seems to not make any difference: the foto taken is always exactly what I see in the preview. I tried 0°, 0° mirrored, 180° and 180° mirrored, always ending up with the same result. Am I getting this wrong? Could you help me out how to have the "natural" (mirrored) view in preview, yet the correct orientation in stills?

mgineer85, 2025-02-15:
Hm, if the cam already embeds some exif orientation data, I am not sure how it behaves in the end. Maybe you just need to turn off the mirroreffect in the camera and let the app handle it?

tomtothet, 2025-02-15:
I just tested 6.0.0b1. The former error is gone. You still want a complete log file? I'll be happy to send one if needed. 

It's a smart idea to give the cam the chance to adapt after changing to high res. With my setup however, it seems that it's not necessary. There's no noticeable difference in the end result between setting it to 0 or 5 for example. If I set 5 however, the preview video disappears quite some time before the image is shot. I would have to set it to 2 to minimize this time, respectively to make sure that you don't even notice it. As not needed with my setup, I will leave the value at 0 however.

In terms of image orientation, there's no difference between 6.0.0a2 and 6.0.0b1. No surprise I guess as you didn't make any changes to that. I tried playing around with the mirroring setting in the UI settings. And this DOES MAKE a difference. The seeting in the backend however does not affect the end result by any means, what ever value I select. This is quite a bummer as a mirrored image is not really a nice thing to have. It's looks even more weird when there's letters in the pic like in my test photos. You think there's a chance to "fix" this? I'm not sure if the Link 2C is causing this. As already mentioned, it seems to mirror the image by default. Unfortunately, I can't change any cam settings on the Pi as the control center doesn't support Linux. I'm quite sure however that this problem is not only with this cam. Maybe it would be necessary to force the mirroring of the pic in the "post-processing", i.e. when the JPEG is generated. A simple toggle switch in the "mediaprocessing" section would do the job :)

I noticed another issue. In the screen when the time counts down before an image is taken, the image section is different to the preview and the actual image taken (those are equal, the one in the count down screen is different). Different means that the section is smaller. I guess I didn't notice it before as the preview and the high res photo used to be the same. With 6.0.0b1 however, the image taken corresponds to the "smaller" image in the count down screen. That's definitely gonna be an issue in real-life settings as it's going to confuse people. People at the very left and right won't be fully visible in the image as anticipated in the preview. Not sure where this behavior is coming from. Any ideas?

mgineer85, 2025-02-15:
> I just tested 6.0.0b1. The former error is gone. You still want a complete log file? I'll be happy to send one if needed.

Doesn't hurt if I could just look at it to see if it goes as intended. Just drop it here if you want.

> It's a smart idea to give the cam the chance to adapt after changing to high res. With my setup however, it seems that it's not necessary. There's no noticeable difference in the end result between [...]

Ok, I saw my camera needs like 3-5 so I set it to 5, maybe I lower the default to 2. Exactly what you perceive is the drawback - no preview while mode changing that is why to keep the number as low as possible.

> In terms of image orientation, there's no difference between 6.0.0a2 and 6.0.0b1. No surprise I guess as you didn't make any changes to that. I tried playing around with the mirroring setting in the [...]

Actually I "force" the orientation in the postprocessing.
If the camera is once configure to spit out "normal" images, non-mirrored on windows, would it continue to so permanently in linux? That would be the easiest option.

If not, to find out whats going on, could you upload an images as it comes from the camera? Just grab a random photo from \media\unprocessed_original folder.

I would have assumed, that the option 0° mirrored should fix it for you. Then in UIsettings enable the mirror effect. This way it should give you a preview as expected by the users (so mirrored, text in the image is flipped) and the still is then non-mirrored (text is readable).

> I noticed another issue. In the screen when the time counts down before an image is taken, the image section is different to the preview and the actual image taken (those are equal, the one in the [...]

Under the condition, that the camera delivers the same field of view for every resolution:
Have you still enabled the frame overlay that is put on top of the image during post-processing?  I mean the one with the photobooth-app logo.  The frame is giving the aspect ratio, not the camera. During post it tries to fit the camera as best as it can to the frame.

tomtothet, 2025-02-15:
> Doesn't hurt if I could just look at it to see if it goes as intended. Just drop it here if you want.

Here's the log file for a fresh start of 6.0.0b1:

[log_20250215_1825.txt](https://github.com/user-attachments/files/18812078/log_20250215_1825.txt)

> Actually I "force" the orientation in the postprocessing.
If the camera is once configure to spit out "normal" images, non-mirrored on windows, would it continue to so permanently in linux? That would be the easiest option.

> If not, to find out whats going on, could you upload an images as it comes from the camera? Just grab a random photo from \media\unprocessed_original folder.

> I would have assumed, that the option 0° mirrored should fix it for you. Then in UIsettings enable the mirror effect. This way it should give you a preview as expected by the users (so mirrored, text in the image is flipped) and the still is then non-mirrored (text is readable).

Sorry, my bad. It apparently needed a hard restart for the settings to take effect. I now activated preview mirroring to give people a natural feeling when adjusting their positions in the image and the option "0° mirrored" in the backend, which gives the end result in the correct orientation.

> Under the condition, that the camera delivers the same field of view for every resolution:
Have you still enabled the frame overlay that is put on top of the image during post-processing? I mean the one with the photobooth-app logo. The frame is giving the aspect ratio, not the camera. During post it tries to fit the camera as best as it can to the frame.

You're probably right. Under uisettings, I deactivated "Enable Livestream Frameoverlay". And the overlay is actually gone in the "idle preview". However, it is still displayed in the "countdown screen". Your explanation makes complete sense and this is probably causing the issue. Do I miss another setting? I looked numerous times, but wasn't able to spot another setting causing the overlay to be displayed during countdown. Can you help me out? If I did actually want to use the overlay, I would have to save it in mirrored format when using the option to mirror the preview. In other words, using the preview mirroring, the overlay is also mirrored. Not sure if you might want to change this behavior.

mgineer85, 2025-02-15:
Oh,  yeah, I see that it can be forgotten easily to restart the service. I add an issue so it can be adressed in the future: #435
Glad that it actually worked ✨

There are many options nowadays to show frame overlay.
1. enable livestream overlay in uisettings -> this is just an overlay for the idle preview, nothing more. It is not embedded in the images in post processing.
2. every action can have its own frame. It is displayed during the countdown and used in post processing.  Try disabling the toggle and see what result you get then. It should have exactly the camera resolution (or at least the aspect ratio in case of thumbnails). The app does not crop anything if not forced to do so by the frame.
[image]
Also you can define different texts per action.
This allows you to have different frame overlayed. Each action has also it's own trigger button 🙈  Quite complex, but once you got it is should be easy.    The documentation might need improvement here :)

mgineer85, 2025-02-16:
#435 is fixed, there is a new toggle to allow restarting ressources on save: 

[image]

mgineer85, 2025-02-16:
I'd consider the original issue for completed now. 
If there are any problems with the implementation or you find room for improvement, please open a new issue.

Thank you for bringing this up and so making the app even more awesome ✨✨✨

closed 2025-02-16 by mgineer85