photobooth-app/photobooth-app #433: Selected filter is not direct shown in the Gallery

Striker62, 2025-02-14:
Selected filter is not direct shown in the Gallery

Go to the Gallery
Select a photo
click filter button
choose a filter

**Expected behavior**
As shown in the video in the docu the picture is shown with the filter
But you have to refresh the browser window to see the effekt

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser Firefox
 - Photobooth 5.1.1

mgineer85, 2025-02-14:
It works for me in chrome. Is this a browser issue maybe?

Striker62, 2025-02-14:
yes , i tested now with chrome. there it works

mgineer85, 2025-02-14:
The cache invalidation seems to work different in Firefox.
Will check it…

mgineer85, 2025-02-15:
seems there is a technique to delete outdated images from cache: https://developer.mozilla.org/en-US/docs/Web/API/Cache/delete

mgineer85, 2025-02-15:
Thanks for bringing this up. It is fixed in v6.
With the changes, whenever an item is updated in the database, all clients are notified and load the new image. If there are no changes, the images are cached in the browser locally now, so the browser would never request an updated image and renders the page faster.

closed 2025-02-15 by mgineer85