photobooth-app/photobooth-app #436: Delete Button: Separate Visibility on Result/Gallery (and centered position)

PixelAlarm, 2025-02-16:
**Describe the bug**
Since the UI overhaul in v5, the delete button can't be separately switched on and off on gallery and result page.
As you disable it in the UI for gallery, it will be gone on result screen as well.

I thought it was meant to be always visible on result screen.
For my taste, I would allow it on result and not allow in gallery. 

A minor cosmetical occurence: When Delete is the only button on result page, it's not centered but shifted to the left.

**To Reproduce**
For Visibility: Toggle Delete button on and off in UI settings.

For not centered view: Disable Filter and Download buttons in UI settings.

**Expected behavior**
Delete could always be activated on result and turned on and off for gallery.

Buttons should be centered, no matter how many are activated.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit, Zorin OS
 - Browser: Chromium, Safari
 - App-Version: v5.1.1 & v6.0.0b1

mgineer85, 2025-02-18:
I tend to enable delete in the result page permanently to allow users deleting bad decisions :)
the other one would be configurable.

mgineer85, 2025-02-18:
itempresenter
[image]

gallery

[image]

mgineer85, 2025-02-18:
ech, that was ugly.
fixed:
[image]

closed 2025-02-18 by mgineer85

tooka, 2025-03-10:
I would like to have a clean preview that is shown after taking the picture (just the image taken, no buttons). The mediaviewer from the gallery shall stay the same (with buttons "back" etc.). 
I tried it with private.css but it seems both the "itempresenter/" and "mediaviewer/" are the same code? is this correct? 
My hope was to find an id that is different in those two, so that i can alter the css code based on that.

mgineer85, 2025-03-10:
You will have the possibility to distinguish in the next release, #471

PixelAlarm, 2025-06-12:
You can disable delete button in Config > UiSettings at the bottom.

Just note, that the button can only be disabled for the gallery and not for the result page (since deleting should always be possible if there is an unpleasant result)

You are still able to hide the button (make it invisible) via private.css

tooka, 2025-06-16:
> > You will have the possibility to distinguish in the next release, [#471](https://github.com/photobooth-app/photobooth-app/issues/471)
> 
> Is this committed anywhere? I have been trying to figure out how to remove delete button from the preview for the past hour lol

You need to check the html code with your browser when the page you want to alter is displayed. Every Browser has an option to check the code. In Firefox it's called "inspector" and you can click on every element in the page and it jumps right to the corresponding code line. From there you can see the stylesheet/css ids that you can then alter using the private.css file. Sometimes you need to check some lines above that for css code that represents the page/route since for example a button can be used on other sites as well and you would want to keep the buttons on those.

For my photobooth-app installation on linux my private.css resides here:

`~/photobooth-data/userdata/private.css`

To get a clean preview after the picture was taken, this is what I added there:
```
/* in itempresenter hide the buttons. works since phootobooth-app version 6.1 */
.route-name-itempresenter .q-pa-md {display: none;} /*del-button*/
.route-name-itempresenter .q-btn {display: none;} /*back-button*/
.route-name-itempresenter .q-icon {display: none;} /*L-R-arrows*/
```

Hope this helps.

srvoets, 2025-06-18:
> You need to check the html code with your browser when the page you want to alter is displayed. Every Browser has an option to check the code. In Firefox it's called "inspector" and you can click on [...]

I managed to get my delete button hidden by using developer mode to find the elements.

But I am going to check out your CSS as well, its likely cleaner.