photobooth-app/photobooth-app #444: Wrong Windows Version shown in admin view under Platform

dtimes, 2025-02-22:
**Is your feature request related to a problem? Please describe.**
first this is a very minor display issue, if you feel that it´s not worth fixing please excuse openingthe issue and just close it

[image]

if running windows11 it´s detected / shown as windows10 under http://localhost:8000/#/admin Platform system
(see screenshot) 

**Describe the solution you'd like**

to my understanding the imported platform does not destinct between windows10 and 11 (same core). I think the fix would be to check for the build number due to windows 11 starting with 22000 (everything above should be win11 not 10) 

simple code example: 

import platform

system = platform.system()
release = platform.release()
version = platform.version()

print(f"System: {system}")
print(f"Release: {release}")
print(f"Version: {version}")

if system == "Windows":
    build_number = int(version.split('.')[2])
    if build_number >= 22000:
        print("You are running Windows 11")
    else:
        print("You are running Windows 10")
else:
    print("You are not running Windows")

output in my case (windows11) 

'System: Windows
Release: 10
Version: 10.0.26100
You are running Windows 11'

**Describe alternatives you've considered**
ignore text in admin interface or change it to "Windows 10 or 11" to avoid confusion

mgineer85, 2025-02-22:
which python Version you use?
there were some fixes in 3.12

dtimes, 2025-02-22:
my bad, I was using Python 3.10.7 (newer version installed but env var / python launcher still pointing to old version )   with  platform.release() resulting in 10  ,  in Python 3.13.2  platform.release() is 11 , the manual already states to use the latest stable python so I guess there is nothing to fix. sorry and thanks for your work

mgineer85, 2025-02-22:
All good. Going to close this, but keep posting issues even if minor to make things better 💯

closed 2025-02-22 by mgineer85