photobooth-app/photobooth-app #445: Blank page because of blocked because of a disallowed MIME type (“text/plain”)

basejumpa, 2025-02-22:
**Describe the bug**
After installation on my Windows 10 computer I get in all browsers (Firefox, Chrome, Edge) a blank page. Developer console says:

[http://localhost:8001/assets/index-BUiL6krN.js”](vscode-file://vscode-app/c:/Users/aman/scoop/apps/vscode-insiders/1.98.0-1739898838051/resources/app/out/vs/code/electron-sandbox/workbench/workbench.html) was blocked because of a disallowed MIME type (“text/plain”).

**To Reproduce**
See above

**Expected behavior**
Make it work.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Windows 10 64bit
 - Browser: Firefox, Chrome, Edge

mgineer85, 2025-02-22:
You use any proxy or something?
The app normally runs on port 8000.
I have never seen this issue before. How did you install it?

mgineer85, 2025-02-22:
on my windows 11 it works just fine. 

i dont think we need to fix it for Windows 10, as it is almost end of life anyways...

[image]

closed 2025-02-24 by mgineer85