photobooth-app/photobooth-app #448: Issues with QR Code Download and PNG Frame

konzertfotografie, 2025-02-26:
Hello,
your app is really great! Unfortunately, i m not a pro, so I cant figure this out on mz own. 
 
Im having trouble downloading the image to my smartphone via QR code. However, when I manually enter the URL in the browser, it works fine.

Also, i m still using the old version. with the new version, my Raspberry Pi 5 didnt recognize the camera ( not sure whats going on there). But even this version is awesome !

On top of that, i seem to be too clueless to properly set up the PNG for the frame - its just not working. 
a lot of problems, sorry :D 

Heres the log of the error:

Traceback (most recent call last):
  File "/home/jsfotografie/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/abstractbackend.py", line 300, in wait_for_hq_image
    return self._wait_for_hq_image()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jsfotografie/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/gphoto2.py", line 161, in _wait_for_hq_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames
2025-02-26 17:09:23,145 [   ERROR] error capture image. attempt=1/3, retrying (abstractbackend.py:303)
2025-02-26 17:09:23,549 [    INFO] (camera_nikon_capture [library.c:4244]) Parentobject of newobject 0x2b000001 is 0x0 now? (port_log.py:20)
2025-02-26 17:09:24,115 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x100b receiving resp failed: PTP Invalid Object Handle (0x2009) (port_log.py:20)
2025-02-26 17:09:24,116 [    INFO] (camera_nikon_capture [library.c:4277]) deleteobject(2b000001) failed: 2009 (port_log.py:20)
2025-02-26 17:09:24,116 [    INFO] (camera_nikon_capture [library.c:4283]) deleteobjectinsdram(2b000001) failed too: 2009 (port_log.py:20)
2025-02-26 17:09:24,118 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property 500c changed, "flashmode" to "Auto" (gphoto2.py:330)
2025-02-26 17:09:24,121 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a4 changed, "liveviewprohibit" to "Liveview should not be prohibited" (gphoto2.py:330)
2025-02-26 17:09:24,123 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1b4 changed, "continousshootingcount" to "63,000000" (gphoto2.py:330)
2025-02-26 17:09:24,125 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property 500e changed, "expprogram" to "M" (gphoto2.py:330)
2025-02-26 17:09:24,126 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property 5013 changed, "capturemode" to "Single Shot" (gphoto2.py:330)
2025-02-26 17:09:24,129 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d0a4 changed, "movieprohibit" to "No movie prohibit conditions" (gphoto2.py:330)
2025-02-26 17:09:24,131 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d0b5 changed, "ISOControlSensitivity" to "1600" (gphoto2.py:330)
2025-02-26 17:09:24,133 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a2 changed, "Live View Status" to "0" (gphoto2.py:330)
2025-02-26 17:09:24,134 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a6 changed, "LiveViewSelector" to "1" (gphoto2.py:330)
2025-02-26 17:09:24,136 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a8 changed, "movieshutterspeed" to "1/60" (gphoto2.py:330)
2025-02-26 17:09:24,138 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a9 changed, "movief-number" to "f/0" (gphoto2.py:330)
2025-02-26 17:09:24,140 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1ab changed, "movieexposurecompensation" to "0" (gphoto2.py:330)
2025-02-26 17:09:24,141 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1af changed, "MovieExposureMeteringMode" to "3" (gphoto2.py:330)
2025-02-26 17:09:24,142 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1b4 changed, "continousshootingcount" to "63,000000" (gphoto2.py:330)
2025-02-26 17:09:24,144 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1f1 changed, "availableshots" to "1578,000000" (gphoto2.py:330)
2025-02-26 17:09:24,146 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1a4 changed, "liveviewprohibit" to "Liveview should not be prohibited" (gphoto2.py:330)
2025-02-26 17:09:24,148 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1b1 changed, "lightmeter" to "-6,000000" (gphoto2.py:330)
2025-02-26 17:09:24,149 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Event 400c, Param1 00010001 (gphoto2.py:330)
2025-02-26 17:09:24,151 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property 5003 changed, "imagesize" to "5568x3712" (gphoto2.py:330)
2025-02-26 17:09:24,153 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property 500a changed, "focusmode" to "AF-A" (gphoto2.py:330)
2025-02-26 17:09:24,154 [   DEBUG] Event: GP_EVENT_CAPTURE_COMPLETE, data: None (gphoto2.py:330)
2025-02-26 17:09:24,166 [   DEBUG] Event: GP_EVENT_UNKNOWN, data: PTP Property d1b4 changed, "continousshootingcount" to "64,000000" (gphoto2.py:330)
2025-02-26 17:09:24,178 [    INFO] got captured_files=[('/', 'capt0000.jpg')] (gphoto2.py:338)
2025-02-26 17:09:24,178 [    INFO] determined file_to_download=('/', 'capt0000.jpg') (gphoto2.py:347)
2025-02-26 17:09:24,193 [    INFO] WledService preset_standby triggered (wledservice.py:132)
2025-02-26 17:09:24,193 [ WARNING] WLED module not connected, ignoring request (wledservice.py:154)
2025-02-26 17:09:24,203 [    INFO] -- process time: 9.06s to capture still (processingservice.py:372)
2025-02-26 17:09:24,203 [    INFO] transition after: counting--(_counted)-->capture (processingservice.py:295)
2025-02-26 17:09:24,204 [    INFO] exit: capture from _captured (processingservice.py:306)
2025-02-26 17:09:24,204 [    INFO] start postprocessing phase 1 (processingservice.py:387)
2025-02-26 17:09:24,205 [    INFO] postprocessing last capture: mediaitem=MediaItem Id: 47f4313e59d31a016caed5a760b30935, filename 20250226-170915-142914.jpg (processingservice.py:391)
2025-02-26 17:09:24,225 [   DEBUG] scaling img by factor 0.25, width input=5568 -> target=1500 (mediaitem.py:532)
2025-02-26 17:09:24,308 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,329 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,349 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,370 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,390 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,399 [   DEBUG] scaling img by factor 0.25, width input=5568 -> target=1200 (mediaitem.py:532)
2025-02-26 17:09:24,411 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,432 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,452 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,473 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,493 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,514 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,535 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-02-26 17:09:24,555 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
[... 82 lines cut ...]
2025-02-26 17:09:25,789 [    INFO] pilgram filter stage ginza to apply (image_pipelinestages.py:22)
2025-02-26 17:09:25,802 [    INFO] 127.0.0.1:36638 - "GET /api/mediaprocessing/preview/47f4313e59d31a016caed5a760b30935/crema HTTP/1.1" 200 (h11_impl.py:476)
2025-02-26 17:09:25,807 [    INFO] 127.0.0.1:36654 - "GET /api/mediaprocessing/preview/47f4313e59d31a016caed5a760b30935/dogpatch HTTP/1.1" 200 (h11_impl.py:476)
2025-02-26 17:09:25,838 [    INFO] pilgram filter stage hefe to apply (image_pipelinestages.py:22)
[... cut ...]
2025-02-26 17:09:41,685 [   ERROR] unknown error occured: error connecting to shareservice dl.php, error 404 {"detail":"Not Found"} (qrshareservice.py:67)
2025-02-26 17:09:51,689 [    INFO] 127.0.0.1:59020 - "GET /media/full/%7Bidetntifier%7D?action=upload_queue HTTP/1.1" 404 (h11_impl.py:476)
2025-02-26 17:09:51,690 [   ERROR] unknown error occured: error connecting to shareservice dl.php, error 404 {"detail":"Not Found"} (qrshareservice.py:67)
2025-02-26 17:09:56,475 [    INFO] getting latest logfile: log/photobooth_20250210.log (debug.py:29)

thanks for your help!
Have a great day!!

mgineer85, 2025-02-26:
Is it just that 7Bidetntifier is spelled wrong?

mgineer85, 2025-02-26:
Regarding the frame, maybe the filename provided was wrong?
It fails to find it.  The file needs to be below the working directory (the directory you start the app in), then userdata -> frames -> endlich34.png.

```
raise FileNotFoundError(f"filepath {str(filepath)} not found!")
FileNotFoundError: filepath userdata/frames/endlich34.png not found!
```

Just tested it with another file here, works as intended.   If you continue to have issues, maybe you can provide additional config and screenshots from your file structure...

mgineer85, 2025-02-26:
Regarding the gphoto issues, it is worth to check google for "gphoto2 {your camera name here}". Other people might have run into issues with the camera.   Usually there is only little we can do on our side but there are many settings in the camera that could be changed...

Also if there is an issue on a new system please check the following service is disabled: https://photobooth-app.org/help/faq/#gphoto2-camera-found-but-no-access

konzertfotografie, 2025-03-05:
Hello, 
I have now corrected the spelling mistake (OMG), but unfortunately, that wasnt the solution.

You mentioned that I should install a publicly accessible PHP host.

I want to use Method B and thought it would be possible to use the "create wifihotspot" function of the raspberry pi 5. i also tried it via my home wifi, of course. However, when loading the QR link in the Browser , i get the Error message: ("detail":"Not found").

Does the PHP service always need to be installed ? 

Just in case, I also ordered a cheap router ( TP-Link TL-WR902AC AC750)...Is this better ?

Her is the errorlog:

 2025-03-05 11:17:39,862 [   DEBUG] photobooth.services.backends.gphoto2 started (gphoto2.py:129)
2025-03-05 11:17:39,863 [    INFO] setting device status from starting to running (abstractbackend.py:122)
2025-03-05 11:17:39,863 [    INFO] device started, set status running (abstractbackend.py:174)
2025-03-05 11:17:39,883 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-05 11:17:39,903 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-05 11:17:39,924 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-05 11:17:39,944 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-05 11:17:39,984 [   DEBUG] configure camera optimized for idle/video (gphoto2.py:205)
2025-03-05 11:17:39,984 [   DEBUG] iso empty, ignore (gphoto2.py:222)
2025-03-05 11:17:39,984 [   DEBUG] shutter speed empty, ignore (gphoto2.py:232)
2025-03-05 11:17:45,032 [    INFO] 10.42.0.1:51454 - "GET /media/processed/full/%7Bidentifier%7D?action=upload_queue HTTP/1.1" 404 (h11_impl.py:476)
2025-03-05 11:17:45,034 [   ERROR] unknown error occured: error connecting to shareservice dl.php, error 404 {"detail":"Not Found"} (qrshareservice.py:67)
2025-03-05 11:17:45,676 [    INFO] trigger action_type='image', action_index=1 (processingservice.py:142)
2025-03-05 11:17:45,677 [    INFO] starting job model: job_model=typ=Typ.image, total_captures_to_take=1, confirmed_captures_collection=[], last_capture=None (processingservice.py:101)
2025-03-05 11:17:45,681 [    INFO] enter: idle from __initial__ (processingservice.py:299)
2025-03-05 11:17:45,681 [    INFO] state idle entered. (processingservice.py:317)
2025-03-05 11:17:45,681 [   DEBUG] starting _process_thread (processingservice.py:118)
2025-03-05 11:17:45,682 [    INFO] starting job (processingservice.py:73)
2025-03-05 11:17:45,682 [    INFO] exit: idle from start (processingservice.py:306)
2025-03-05 11:17:45,683 [    INFO] enter: counting from start (processingservice.py:299)
2025-03-05 11:17:45,684 [    INFO] 127.0.0.1:58336 - "GET /api/actions/image/1 HTTP/1.1" 200 (h11_impl.py:476)
2025-03-05 11:17:45,685 [    INFO] WledService preset_thrill triggered (wledservice.py:137)
2025-03-05 11:17:45,685 [ WARNING] WLED module not connected, ignoring request (wledservice.py:154)
2025-03-05 11:17:45,686 [    INFO] started countdown, duration=5.0, offset_camera=0.2 (processingservice.py:333)
2025-03-05 11:17:45,690 [   DEBUG] configure camera optimized for idle/video (gphoto2.py:205)
2025-03-05 11:17:45,692 [   DEBUG] iso empty, ignore (gphoto2.py:222)
2025-03-05 11:17:45,692 [   DEBUG] shutter speed empty, ignore (gphoto2.py:232)
2025-03-05 11:17:50,491 [    INFO] transition after: idle--(start)-->counting (processingservice.py:295)
2025-03-05 11:17:50,492 [    INFO] exit: counting from _counted (processingservice.py:306)
2025-03-05 11:17:50,492 [    INFO] enter: capture from _counted (processingservice.py:299)
2025-03-05 11:17:50,493 [    INFO] current capture (1/1, remaining 0) (processingservice.py:346)
2025-03-05 11:17:50,496 [   DEBUG] capture to mediaitem.path_original=PosixPath('media/original/20250305-111750-493977.jpg') (processingservice.py:358)
2025-03-05 11:17:50,497 [    INFO] WledService preset_shoot triggered (wledservice.py:142)
2025-03-05 11:17:50,498 [ WARNING] WLED module not connected, ignoring request (wledservice.py:154)
2025-03-05 11:17:50,501 [    INFO] disable viewfinder before capture (gphoto2.py:305)
2025-03-05 11:17:50,502 [   DEBUG] request to _wait_for_lores_image waiting until ongoing request_hires_still is finished (gphoto2.py:174)
2025-03-05 11:17:50,516 [   DEBUG] request to _wait_for_lores_image waiting until ongoing request_hires_still is finished (gphoto2.py:174)
2025-03-05 11:17:50,532 [    INFO] (_get_config [config.c:11188]) Type of property 'Lens Name' expected: 0x0002 got: 0x0004 (port_log.py:20)
2025-03-05 11:17:50,533 [    INFO] (_get_config [config.c:11204]) Widget get of property 'Lens Name' failed, trying to see if we have another... (port_log.py:20)
2025-03-05 11:17:50,589 [    INFO] (_get_config [config.c:11188]) Type of property 'Exposure Program' expected: 0x0002 got: 0xffff (port_log.py:20)
2025-03-05 11:17:50,604 [    INFO] (_get_config [config.c:11188]) Type of property 'Live View AF Mode' expected: 0x0001 got: 0x0002 (port_log.py:20)
2025-03-05 11:17:50,604 [    INFO] (_get_config [config.c:11204]) Widget get of property 'Live View AF Mode' failed, trying to see if we have another... (port_log.py:20)
[... cut ...]
2025-03-05 11:17:56,980 [    INFO] image_frame_stage to apply (image_pipelinestages.py:215)
2025-03-05 11:17:56,980 [   ERROR] filepath userdata/frames/endlich34.png not found! (image_pipelinestages.py:224)

i have that set under QR share in the settings under ShareService URL

http://10.42.0.1:8000/media/processed/full/{identifier}

and this is what appears in the browser when i open the link

10.42.0.1:8000/media/processed/full/%7Bidentifier%7D?action=download&id=1ae1cdc4b5ffa8db-fc3134fcc0a9a863

Thank you for your help ! 
And please forgive my lack of knowledge :D .

mgineer85, 2025-03-05:
for v4 it would be:
```
http://10.42.0.1:8000/media/processed/full/{filename}
```

for v5 and above:
http://10.42.0.1:8000/media/full/{identifier}

Also apologies for the misleading configuration, but for method B you would need to DISABLE the qr share service.
I will improve the docs/configuration to avoid confusion in the future...

Works?

konzertfotografie, 2025-03-06:
Hello,
Thanks for the quick response!
 I copied an pasted your settings. Additionally, i "DISABLED" the first switch under QR SHARE ENABLED  ( turned off the blue switch)

Now, after triggering the camera, the app doesnt even open the photo preview anymore.

Sorry, but unfortunately, this isnt the solution either.

ive also tried reinstalling multipe times. I always end up at this same point, as if i keep making the same mistake somewhere (:D )

Heres the log:

2025-03-06 10:27:38,844 [   DEBUG] scaling img by factor 0.25, width input=5568 -> target=1500 (mediaitem.py:532)
2025-03-06 10:27:38,947 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:38,968 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:38,988 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,009 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,025 [   DEBUG] scaling img by factor 0.25, width input=5568 -> target=1200 (mediaitem.py:532)
2025-03-06 10:27:39,030 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,050 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,071 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,092 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,113 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,134 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,155 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,175 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,196 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,208 [   DEBUG] scaling img by factor 0.125, width input=5568 -> target=400 (mediaitem.py:532)
2025-03-06 10:27:39,216 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,239 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,260 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,280 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,301 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,321 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,342 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,354 [    INFO] -- process time: 0.52s to create scaled images (processingservice.py:396)
2025-03-06 10:27:39,354 [    INFO] image_img_background_stage to apply, background_file='backgrounds/pink-7761356_1920.jpg' reverse=False (image_pipelinestages.py:172)
2025-03-06 10:27:39,354 [ WARNING] no transparency in image, background stage makes no sense to apply! (image_pipelinestages.py:175)
2025-03-06 10:27:39,354 [    INFO] image_frame_stage to apply (image_pipelinestages.py:215)
2025-03-06 10:27:39,363 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2025-03-06 10:27:39,415 [    INFO] loaded frame_path=PosixPath('/home/jsfotografie/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/assets/frames/pixabay-holidays-1798208_1920.png') (image_pipelinestages.py:227)
2025-03-06 10:27:39,429 [    INFO] detected transparent area transparent_xy=(0, 0, 1920, 1371), transparent_size=(1920, 1371) (image_pipelinestages.py:240)
2025-03-06 10:27:39,506 [    INFO] captured image fitted to image_fitted.size=(1920, 1371), original capture was image.size=(1392, 928) (image_pipelinestages.py:252)
2025-03-06 10:27:39,507 [   DEBUG] ratio_original=0.6666666666666666, ratio_fitted=0.7140625 (image_pipelinestages.py:253)
2025-03-06 10:27:39,537 [    INFO] text stage to apply (image_pipelinestages.py:50)
2025-03-06 10:27:39,538 [   DEBUG] apply text: textconfig=TextsConfig(text='Made with the photobooth-app', pos_x=100, pos_y=1300, rotate=0, font_size=40, font='fonts/Roboto-Bold.ttf', color=Color('#ccc', rgb=(204, 204, 204))) (image_pipelinestages.py:53)
2025-03-06 10:27:39,556 [    INFO] 3 stages applied in: 0.2s (mediaprocessingservice.py:187)
[... cut ...]
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/home/jsfotografie/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/mediacollection/mediaitem.py", line 209, in share_url
    return appconfig.qrshare.share_custom_qr_url.format(filename=self.filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'identifier'
2025-03-06 10:28:42,223 [    INFO] getting latest logfile: log/photobooth_20250306.log (debug.py:29)

konzertfotografie, 2025-03-06:
OOOOOH !!! 

When I use your pre-configured button, it works !!!  WUUUHUUU !!! :D 

Now I just need to figure out the other issues and see whats going on :D 

I'll take a look at the frames now!

(omg, the error is often sitting in front of the screen...Shame on me)  :D 
Well, at least its working now :D 

see you in a bit  (frames...problem )

thanks !!

konzertfotografie, 2025-03-06:
OKAY !!! IT WORKS !!! 

Thank you !!

Hmmm, I think its running so well right now that im considering creating a backup of everything and trying out V5 :D 

Many, many thanks for your help and patience !

Small not:
when youre in the "status" section under "System Information", I can conveniently open the file directory via a button, which is realy great!
Could you add such a button in the "configuratin-Action-Image" section as well, along with a note like "create Frames folder here" ? :D I initally created the folder in the wrong place becuse i was overthinking it! maybe this could help others as well ? :D 

Alright, thats it for me today.

Really amazing support from you!! 
Super app !!
I love finding projects like this on the internet :)
Thanks for Your work !

I'll take another look at v5 in the next few days. As i mentioned, the Raspberry 5 didnt recognize the camera there :D 

See you soon :D 
(okay?? :p )

mgineer85, 2025-03-06:
Okay! Great!
Maybe you want to go to v6 directly then? 🤣

closed 2025-03-06 by mgineer85